spencer-project/spencer_people_tracking #2: Cannot find spencer_tracking_msgs/DetectedPersons.h while compiling

Po-Jen, 2015-07-06:
Hi, I am using ROS Indigo with Ubuntu 14.04 (64-bit). 

I run these steps:

    cd ~catkin_ws/src
    git clone https://github.com/spencer-project/spencer_people_tracking.git
    cd ..
    catkin_make

But I get the following errors:

     Scanning dependencies of target spencer_detected_person_association
    [ 57%] Building CXX object spencer_people_tracking-master/detection/spencer_detected_person_association/CMakeFiles/spencer_detected_person_association.dir/src/spencer_detected_person_association/convert_to_composite_detections.cpp.o
    In file included from /home/iceira/ws/src/spencer_people_tracking-master/detection/spencer_detected_person_association/src/spencer_detected_person_association/convert_to_composite_detections.cpp:2:0:
    /home/iceira/ws/src/spencer_people_tracking-master/detection/spencer_detected_person_association/src/spencer_detected_person_association/convert_to_composite_detections.h:11:51: fatal error: spencer_tracking_msgs/DetectedPersons.h: No such file or directory
     #include <spencer_tracking_msgs/DetectedPersons.h>
                                                       ^
    compilation terminated.
    make[2]: *** [spencer_people_tracking-master/detection/spencer_detected_person_association/CMakeFiles/spencer_detected_person_association.dir/src/spencer_detected_person_association/convert_to_composite_detections.cpp.o] Error 1
    make[1]: *** [spencer_people_tracking-master/detection/spencer_detected_person_association/CMakeFiles/spencer_detected_person_association.dir/all] Error 2
    make[1]: *** Waiting for unfinished jobs....
    Linking CXX executable /home/iceira/ws/devel/lib/pcl_people_detector/detect_people
    [ 57%] Built target detect_people
    make: *** [all] Error 2
    Invoking "make -j2 -l2" failed

I tried to find DetectedPersons.h by locate command, and found it is only in 

    ~/catkin_ws/devel/include/spencer_tracking_msgs/

Not sure what cause this problem, any help are appreciated. Thanks.

makokal, 2015-07-26:
@tlind did someone already take care of this somehow?

tlind, 2015-08-06:
Thanks for the reminder @makokal. This has hopefully been fixed now by cb59168e077. The `CMakeLists.txt` was missing `${catkin_INCLUDE_DIRS}`. Not sure why we did not experience the issue on our machines.

makokal, 2015-08-06:
Shall we close this issue then?

closed 2015-08-06 by tlind