spencer-project/spencer_people_tracking #6: Problem with spencer project

hacker-boy, 2016-05-10:
Hello,
I'm working on project and i have found this awesome project witch inspired me alot so i went to run but i have some problems first i have kinect for xbox so openni2 won't work so i really want to run just the detected persons and tracked peoples so please is any one can help me i'm stuck and this is my master project i need to read data from detected&tracked nodes to apply a classifier on them please help me :(

tlind, 2016-05-10:
You mean the old Kinect v1 for Xbox 360? Or the newerKinect v2 for the Xbox One?

hacker-boy, 2016-05-10:
i hev the old one kinect for xbox 360

hacker-boy, 2016-05-10:
tlind please help

hacker-boy, 2016-05-10:
tlind i have tried your peopledetector but i think its not for catkin

Sifouu, 2016-05-10:
in the launch file try to change openni2 to openni

hacker-boy, 2016-05-10:
i did but nothing happed i have deleted it complitly from the launch

hacker-boy, 2016-05-10:
the thing that i want to know is ; spencer works with kinect v1 or not

Sifouu, 2016-05-10:
you can read in the (Tutorial 1: People / group tracking and visualization with a single RGB-D sensor)

......Place your Asus Xtion Pro Live or Kinect v1 sensor horizontally on a flat surface.....

so i think YES it works

hacker-boy, 2016-05-10:
i wiil try it now

hacker-boy, 2016-05-10:
so i have tried it but nothing show on rviz the camera-rgbd is disabled why ???

tlind, 2016-05-10:
Yes Kinect v1 or Asus Xtion Pro Live definitely works. Switching to openni2 was necessary for the newer Asus Xtion Pro Live firmware (with USB 3 support) to work. I assume that for the MS Kinect v1, openni (the original version) still works. Just reply openni2 with openni and openni2.launch with openni.launch, and make sure you have it installed of course.

You can enable the camera display if you want to. It is disabled just to not clutter the screen.
The first step is to ensure you get RGB and depth data on the corresponding ROS topics (e.g. using RViz), and then check if the people detector is working.

hacker-boy, 2016-05-10:
its showing now the pcl but rgbdcamera cant be enable it's still disable

Sifouu, 2016-05-10:
Tlind,  i did all what you said ( openni2 ----> openni) and all the required installation correctly 
Cuda, PCL ....etc. 
but when i run " tracking_single_rgbd_sensor.launch " with Kinect v1 and i getted the RGBD data 

but the people detector doesn't work.

Please i need your help it is also for my Master project.

hacker-boy, 2016-05-10:
siffou  i getted the RGBD data how in rviz or topic or msg

Sifouu, 2016-05-10:
in rviz . (the launch file will run rviz with a predefined configuration)

hacker-boy, 2016-05-10:
yes

hacker-boy, 2016-05-10:
Tlind every thing works fine just it does not track the person no tracking no detection just i see topics is there any thing i miss , is it going to draw some rectangle in rviz

Sifouu, 2016-05-10:
The same problem with me
i tried to see the detected_persons Topic 
by:
rostopic echo /spencer/perception/detected_persons
i had this error 
ERROR: Cannot load message class for [spencer_tracking_msgs/DetectedPersons]. Are your messages built?

???

hacker-boy, 2016-05-10:
sifo dir source devel/setup.bash

Sifouu, 2016-05-10:
Larbi mchalek , rahou idetecti ???

hacker-boy, 2016-05-10:
non makash detection sifou ta3 gsi

closed 2016-08-04 by tlind