spencer-project/spencer_people_tracking #7: ROS package runs correctly but within any  functionality ( Detection or tracking ..)

Sifouu, 2016-05-11:
Hi, i'm using Ubuntu 14.04 (64 bit), ROS indigo, and Kinect V1
1- so i did all these steps 
Installing Openni
  //           Cuda 7.5 
 //             PCL
changing Openni2 ---> Openni in the launch file
...etc.
2- I compile the package correctly. 
3- i got the RGB and depth data on the corresponding ROS topics and Rviz

but when i launch " roslaunch spencer_people_tracking_launch tracking_single_rgbd_sensor.launch height_above_ground:=1.6"

 There is no detection or tracking 

So i want to know if there are other configurations to do, or dependencies to install.
and does the  height of the camera above the ground influence in the detection ????

 Thank you

hacker-boy, 2016-05-11:
same problem like yours to do the detection and tracking i'm using openni_tracker i really want if spencer work

tlind, 2016-05-11:
A first step to pinpoint the issue would be to check the if you receive ROS messages (at least empty ones) on the corresponding DetectedPersons and TrackedPersons topics. I unfortunately cannot try this myself right now, but I think the topics should be

    /spencer/perception/detected_persons
    /spencer/perception/tracked_persons

and you can use `rostopic echo` on these topics once you have started the launch file.

If you receive no detected persons, then there is an issue with the detector, and I would not worry about the tracker yet.

You can also do a `rosnode list` to find the name of the detector (could be something like `/spencer/perception_internal/detected_persons/upper_body` or so), and then do a `rosnode info ` followed by a node name, to see if any required input topics are missing (i.e. incorrectly mapped). It could be that with openni 1, the names of some RGB and depth topics are slightly different (e.g. `image` instead of `image_color` or so).

Sifouu, 2016-05-19:
Hi Timm Thanks for your response
I receive an empty messages
and i didn't understand your issue for rosnode info please can you clarify please

Thank you

tlind, 2016-06-22:
Unknown type means that these topics are not being published. Check if there are similar topics published by your RGB-D driver, which can be remapped to these in a roslaunch file.

tlind, 2017-03-22:
This should be fixed now by #20.

closed 2017-03-22 by tlind