spencer-project/spencer_people_tracking #8: trying to make spencer work from a bag file.

Garmar, 2016-06-11:
I have a bag fille from a rgbd camera, and I'm trying to follow the steps in the tutorial 1, 
I've created a launch file to set the sim_time true and to remap the topic's name.

<launch>

<param name="use_sim_time" value="true"/>

<node pkg="rosbag" type="play" name="player" output="screen" args="--clock /home/gloria/camera_2016-05-30-16-29-34.bag /camera/depth_registered/image_raw:=/spencer/sensors/rgbd_front_top/depth_registered/image_raw /camera/rgb/camera_info:=/spencer/sensors/rgbd_front_top/rgb/camera_info /camera/rgb/image_raw:=/spencer/sensors/rgbd_front_top/rgb/image_raw /camera/depth_registered/camera_info:=/spencer/sensors/rgbd_front_top/depth_registered/camera_info"/>

</launch>

I've also modified the value of openni2 load_driver to false in the launch file,
then I launch de bag file before doing:

roslaunch spencer_people_tracking_launch tracking_single_rgbd_sensor.launch height_above_ground:=1.6

when I do rostopic hz /spencer/perception/detected_persons

all I get is:
WARNING: may be using simulated time

I can display the data in the bag doing rosrun image_view image_view image:= topic name. and it does it properly, so the data in the bag is properly recorded.

Is there any thing I can check or do to make this work?
thak you very much

tlind, 2016-06-13:
You can try to list all detector nodes using `rosnode list /spencer/perception_internal/people_detection`. Then, for each one, you can type `rosnode info ` followed by the fully-qualified node name.

This will give you a list of all ROS topics that the node subscribes to. Check, especially for the sensor topics, if they are subscribed properly. Probably, this is a mismapping of topic names. E.g.:

    Node [/spencer/perception_internal/people_detection/rgbd_front_top/upper_body_detector]
    ...
     * /spencer/sensors/rgbd_front_top/rgb/image_raw [unknown type]
     * /spencer/sensors/rgbd_front_top/rgb/camera_info [unknown type]
     * /spencer/perception_internal/rwth_ground_plane [rwth_perception_people_msgs/GroundPlane]
     * /spencer/sensors/rgbd_front_top/depth/image_rect [unknown type]

Here you can see that 3 topics are not properly subscribed because it says "unknown type".

tlind, 2017-03-22:
As documented in #27, I have just added an example bagfile along with instructions in the README (see "Tutorials" section).

closed 2017-03-22 by tlind