spencer-project/spencer_people_tracking #9: Unable to make and install libcudaHOG lib

srinimd2005, 2016-06-14:
Hi eveyone,
I couldn't make and install the libcudaHOGlib even though I have cuda in my system. I even tried the cuda samples and it is working. The cmake fails in cudaHOG.cpp build. 

Is the groundHOG necessary for people_tracking or can I even try using only rgbd_people_tracking ?

Can I use Kinect from xbox 360. The rviz display pointcloud but no human tracking. 

Please someone help on this issue.

srinimd2005, 2016-06-17:
Hi I managed to run the spencer tracking with some more modification. But when the tracking is done sometimes I get this error. 
Can someone please help me to resolve this. This happens only for upper_body_detection and not for HOG based tracking.

*** Error in `/home/ezc2x-ros/cat_ws/devel/lib/rwth_upper_body_detector/upper_body_detector': double free or corruption (!prev): 0x0000000002815770 ***
[spencer/perception_internal/people_detection/rgbd_front_top/upper_body_detector-32] process has died [pid 18968, exit code -6, cmd /home/ezc2x-ros/cat_ws/devel/lib/rwth_upper_body_detector/upper_body_detector __name:=upper_body_detector __log:=/home/ezc2x-ros/.ros/log/e03e4c96-347d-11e6-a1de-0022156bbb6b/spencer-perception_internal-people_detection-rgbd_front_top-upper_body_detector-32.log].
log file: /home/ezc2x-ros/.ros/log/e03e4c96-347d-11e6-a1de-0022156bbb6b/spencer-perception_internal-people_detection-rgbd_front_top-upper_body_detector-32*.log

srinimd2005, 2016-06-17:
The problem could be solved by adding right hyperlink to the make file. The weblink to the file has been changed to another destination.

http://www.vision.rwth-aachen.de/media/resource_files/groundHOG_v1.1_b.tar.gz
in this file  /spencer_people_tracking/detection/monocular_detectors/3rd_party/CMakeLists.txt

closed 2016-06-17 by srinimd2005

Alex-Beh, 2020-06-16:
Hello, I encounter some issue when I want to compile the libcudaHOG lib.

when I run `sudo make` in the build directory, the following error pop out:
```
../../libcudaHOG/cudaHOG/gradients.cu(150): warning: a user-provided literal suffix must begin with "_"
../../libcudaHOG/cudaHOG/gradients.cu(150): error: user-defined literal operator not found
```
Currently I try to comment those line and continue to compile it... (although I know it is not a good solution in long run.)

Then I face the `-lboost_program_options-mt`, I change according what you said in [url](https://github.com/spencer-project/spencer_people_tracking/tree/master/detection/monocular_detectors/3rd_party#important), but the following error pop up:
```
/usr/bin/ld: main.o: undefined reference to symbol '_ZN7QString4freeEPNS_4DataE'
//usr/lib/x86_64-linux-gnu/libQtCore.so.4: error adding symbols: DSO missing from command line
collect2: error: ld returned 1 exit status
```
Something wrong when I include the library, do you have any idea on this?

Tacha-S, 2020-07-17:
@Alex-Beh I was occured a same error, how did you solve it?

> ```
> /usr/bin/ld: main.o: undefined reference to symbol '_ZN7QString4freeEPNS_4DataE'
> //usr/lib/x86_64-linux-gnu/libQtCore.so.4: error adding symbols: DSO missing from command line
> collect2: error: ld returned 1 exit status
> ```

Alex-Beh, 2020-07-17:
Follow the instructions in **IMPORTANT!** in [url](https://github.com/spencer-project/spencer_people_tracking/tree/master/detection/monocular_detectors/3rd_party#important).

Let me know if you still cannot solve it.

Tacha-S, 2020-07-20:
Solved the error, Thanks.
'_ZN7QString4freeEPNS_4DataE' shows `QString::fromAscii_helper` reference