spencer-project/spencer_people_tracking #11: Queries in installing libcudaHOG

sfchik, 2016-06-19:
Hi,

I am interested in using the `spencer_people_tracking` in my master project, and hence I attempted to install the libcudaHOG from `spencer_people_tracking/detection/monocular_detectors/3rd_party/`. I have installed Nvdia-Cuda 6.5 and try to follow the installation instruction. The first issue that I encountered is with the statement "Change into the rwth_perception_people/3rd_party directory", may I know does that mean "/<my workspace>/src/spencer_people_tracking/detection/monocular_detectors/3rd_party"? 

Then I try the next instructions "mkdir build; cd build; cmake ..; make" The followings was shown in the terminal:

-- The C compiler identification is GNU 4.8.4
-- The CXX compiler identification is GNU 4.8.4
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Preparing libcudaHOG
-- Created pkg-config file: /home/aimachine/catkin_ai/src/spencer_people_tracking/detection/monocular_detectors/3rd_party/build/libcudaHOG/pkgconfig/cudaHOG.pc
-- Will download libcudaHOG from: http://www.vision.rwth-aachen.de/projects/groundHOG_v1.1_b.tar.gz
-- Configuring done
-- Generating done
-- Build files have been written to: /home/aimachine/catkin_ai/src/spencer_people_tracking/detection/monocular_detectors/3rd_party/build

Then I run sudo make install and get some errors:

Scanning dependencies of target libcudaHOG
[ 12%] Creating directories for 'libcudaHOG'
[ 25%] Performing download step (download, verify and extract) for 'libcudaHOG'
-- downloading...
     src='http://www.vision.rwth-aachen.de/projects/groundHOG_v1.1_b.tar.gz'
     dst='/home/aimachine/catkin_ai/src/spencer_people_tracking/detection/monocular_detectors/3rd_party/build/libcudaHOG/src/groundHOG_v1.1_b.tar.gz'
     timeout='none'
CMake Error at libcudaHOG-stamp/download-libcudaHOG.cmake:22 (message):
  error: downloading
  'http://www.vision.rwth-aachen.de/projects/groundHOG_v1.1_b.tar.gz' failed

    status_code: 22
    status_string: "HTTP response code said error"
    log: Hostname was NOT found in DNS cache
    Trying 137.226.161.13...

  Connected to www.vision.rwth-aachen.de (137.226.161.13) port 80 (#0)

  GET /projects/groundHOG_v1.1_b.tar.gz HTTP/1.1

  User-Agent: curl/7.35.0

  Host: www.vision.rwth-aachen.de

  Accept: */*

  The requested URL returned error: 404 NOT FOUND

  Closing connection 0

make[2]: *** [libcudaHOG/src/libcudaHOG-stamp/libcudaHOG-download] Error 1
make[1]: *** [CMakeFiles/libcudaHOG.dir/all] Error 2
make: *** [all] Error 2
  
May I know does this mean that the library link is broken? Or I have missed out something? Thank you.

tlind, 2016-06-22:
The issue appears because the URL for the download has changed recently. See #9. We will fix the URL in the next release.

sfchik, 2016-06-25:
Hi tlind, thanks for your reply. I will try it out. Appreciate.

tlind, 2016-08-04:
Download URL has been fixed in the latest version.

closed 2016-08-04 by tlind