spencer-project/spencer_people_tracking #16: Detection working but tracking is not

pirobot, 2016-08-28:
Hello,

I'm trying out the spencer_people_tracking package using ROS Indigo, Ubuntu 14.04 and an Xtion Pro Live camera.  I am not using the CUDA Hog detection since I don't have a CUDA-capable GPU.

The source code compiled without error and running 'roslaunch spencer_people_tracking_launch tracking_on_robot.launch' connects to the camera and launches all nodes without error.

However in RViz, when I stand and move in front of the camera, I am detected OK and the bounding box moves with my movement when using the "Fused detections" visualizer.  However, no motion tracks are visualized even though I have the "TrackedPersons Visual Confirmed" box checked and the topic set to /spencer/perception/tracked_persons.  If I run 'rostopic echo /spencer/perception/tracked_persons', no data appears when I move in front of the camera.

tlind, 2017-03-13:
Sorry for the late reply; hard to guess in hindsight what went wrong in your case. In general, the tracker only initiates tracks if the detection is sufficiently stable (i.e. doesn't disappear every 2nd frame or so). Also, if you have a map published on the /map topic, no tracks are initiated in map cells that are marked as occupied. 
This is under the assumption that your `rostopic echo /spencer/perception/tracked_persons` call outputs empty TrackedPersons messages. If you receive no messages at all, something else is wrong.

Closing the issue because it was opened quite a while ago.

closed 2017-03-13 by tlind

pirobot, 2017-03-14:
Thanks--I tried it again the other day and now it appears to be working.  Go figure!