spencer-project/spencer_people_tracking #17: Trying to use PrimeSense Carmine 1.08/1.09 instead of Asus Xtion Pro or Kinect v1

sfchik, 2016-09-26:
Hi,

I finally successfully built the spencer_people_tracking package. But I current have only kinect v1, which I tried to implement with, but it seems like openni2 is unable to recognize kinect v1. Is it because openni2 does not support kinect as stated in the package page:

 "This package contains launch files for using OpenNI-compliant devices in ROS. It supports the Asus Xtion, Xtion Pro, and multiple version of the Primesense 1.08 and 1.09 cameras. It does NOT support any versions of the Kinect. freenect_launch or openni_launch is the recommended package for using a Kinect with ROS. "

Hence, I tried changing  <include file="$(find openni2_launch)/launch/openni2.launch"> to  <include file="$(find openni_launch)/launch/openni.launch"> in the tracking_single_rgbd_sensor.launch and it seems like it is able to run the camera but does not do neither detection nor tracking.

Is there any solution I can run the package using Kinect v1, or I have to use Asus Xtion Pro which currently the production of the camera seems discontinued, or I can use PrimeSense Carmine 1.08/1.09 which is still available in the market. 

Thanks in advance.

tlind, 2017-03-08:
Hi! I'm totally sorry for not being able to reply to your comment earlier. Our lab doesn't exist anymore and I can only give support here in my free time.  Switching to OpenNI 1 is certainly possible, because we also used it initially in our experiments. We only switched over to OpenNI2 at some point because of USB 3-related issues.

The trick is to remap the input topics that the upper-body and groundHOG detectors subscribe to. IIRC, for example, OpenNi2 publishes a topic called `image_color`, but in OpenNi1 it was called `image` only. I will try to provide a second launch file for OpenNi1 and update the documentation accordingly as soon as I find the time.

sfchik, 2017-03-09:
Hi tlind,

It's okay, no problem. I purchased an  Asus Xtion Pro, and it's working fine. But I have some hard time combining the laser leg detector and the camera data, I would like to open another issue and hope to get some guidances. Thanks for your reply! XD

tlind, 2017-03-13:
Ok, I'll close this issue then. For further discussion/progress of Kinect v1 support, see #20.

closed 2017-03-13 by tlind