spencer-project/spencer_people_tracking #19: srl_nearest_neighbor_tracker-process crashes

femtogram, 2017-03-07:
When trying to fuse RGBD and LIDAR, srl_nearest_neighbor_tracker-process crashes:

```
nnt_node: /usr/include/boost/smart_ptr/shared_ptr.hpp:648: typename boost::detail::sp_member_access<T>::type boost::shared_ptr<T>::operator->() const [with T = const spencer_tracking_msgs::DetectedPersons_<std::allocator<void> >; typename boost::detail::sp_member_access<T>::type = const spencer_tracking_msgs::DetectedPersons_<std::allocator<void> >*]: Assertion `px != 0' failed.
[spencer/perception_internal/people_tracking/srl_nearest_neighbor_tracker-55] process has died [pid 3075, exit code -6, cmd /home/kota/spencer_kinetic_ws/devel/lib/srl_nearest_neighbor_tracker/nnt_node __name:=srl_nearest_neighbor_tracker __log:=/home/kota/.ros/log/b2c33872-034d-11e7-bd4e-64006a58bea8/spencer-perception_internal-people_tracking-srl_nearest_neighbor_tracker-55.log].
log file: /home/kota/.ros/log/b2c33872-034d-11e7-bd4e-64006a58bea8/spencer-perception_internal-people_tracking-srl_nearest_neighbor_tracker-55*.log
[spencer/perception_internal/people_detection/rgbd_front_top/upper_body_detector-31] process has died [pid 2727, exit code -11, cmd /home/kota/spencer_kinetic_ws/devel/lib/rwth_upper_body_detector/upper_body_detector __name:=upper_body_detector __log:=/home/kota/.ros/log/b2c33872-034d-11e7-bd4e-64006a58bea8/spencer-perception_internal-people_detection-rgbd_front_top-upper_body_detector-31.log].
log file: /home/kota/.ros/log/b2c33872-034d-11e7-bd4e-64006a58bea8/spencer-perception_internal-people_detection-rgbd_front_top-upper_body_detector-31*.log
```

The log file is empty.
The current workaround I have been using is to start up srl_nearest_neighbor_tracker manually:

```
rosrun srl_nearest_neighbor_tracker nnt_node __name:=srl_nearest_neighbor_tracker
```

I'm on Ubuntu 16.04 x86_64 with ROS Kinetic.
Thanks

tlind, 2017-03-08:
Could you provide a GDB stacktrace (by adding e.g. `launch-prefix='xterm -e gdb -ex run --args`) to `srl_nearest_neighbor_tracker/launch/nnt.launch`? Hard to pinpoint otherwise.

femtogram, 2017-03-09:
Oh man, sorry about that, I totally forgot to post the backtrace didn't I.

Here is the output of the program up to the crash:

```
[ INFO] [1489078593.943810111]: Deleting 0 duplicated tracks.
[ INFO] [1489078593.943936021]: Publishing 2 tracked persons!
[ INFO] [1489078593.974179852]: Received 35 observations.
[ INFO] [1489078593.975275551]: Occlusion manager returned 0 tracks
[ INFO] [1489078593.981312140]: Number of accepted initation candidates 75

[ INFO] [1489078593.981382595]: Deleting 0 duplicated tracks.
[ INFO] [1489078593.981484812]: Publishing 2 tracked persons!
[ INFO] [1489078593.984951106]: Received 36 observations.
[ INFO] [1489078593.985936310]: Occlusion manager returned 0 tracks
[ INFO] [1489078593.993183581]: Number of accepted initation candidates 67

[ INFO] [1489078593.993285936]: Deleting 0 duplicated tracks.
[ INFO] [1489078593.993390776]: Publishing 2 tracked persons!
[ INFO] [1489078594.018606311]: Received 38 observations.
[ INFO] [1489078594.019371343]: Occlusion manager returned 0 tracks
[ INFO] [1489078594.026633078]: Number of accepted initation candidates 71

[ INFO] [1489078594.026734576]: Deleting 0 duplicated tracks.
[ INFO] [1489078594.026824553]: Publishing 2 tracked persons!
[ INFO] [1489078594.062235149]: Received 34 observations.
[ INFO] [1489078594.063062375]: Occlusion manager returned 0 tracks
[ INFO] [1489078594.069363759]: Number of accepted initation candidates 69

[ INFO] [1489078594.069461929]: Deleting 0 duplicated tracks.
[ INFO] [1489078594.069567768]: Publishing 2 tracked persons!
[ INFO] [1489078594.073102667]: Received 36 observations.
[ INFO] [1489078594.073883891]: Occlusion manager returned 0 tracks
[ INFO] [1489078594.080420101]: Number of accepted initation candidates 70

[ INFO] [1489078594.080515548]: Deleting 0 duplicated tracks.
[ INFO] [1489078594.080630265]: Publishing 2 tracked persons!
[ INFO] [1489078594.096228224]: Received 34 observations.
[ INFO] [1489078594.097061389]: Occlusion manager returned 0 tracks
[ INFO] [1489078594.102842329]: Number of accepted initation candidates 69

[ INFO] [1489078594.102918925]: Deleting 0 duplicated tracks.
[ INFO] [1489078594.103053065]: Publishing 2 tracked persons!
[ INFO] [1489078594.130234309]: Received 33 observations.
[ INFO] [1489078594.131170087]: Occlusion manager returned 0 tracks
[ INFO] [1489078594.137249535]: Number of accepted initation candidates 65

[ INFO] [1489078594.137321892]: Deleting 0 duplicated tracks.
[ INFO] [1489078594.137419605]: Publishing 2 tracked persons!
[ INFO] [1489078594.190050428]: Received 39 observations.
[ INFO] [1489078594.191098808]: Occlusion manager returned 0 tracks
[ INFO] [1489078594.198209911]: Number of accepted initation candidates 68

[ INFO] [1489078594.198300735]: Deleting 0 duplicated tracks.
[ INFO] [1489078594.198399955]: Publishing 2 tracked persons!
[ INFO] [1489078594.201453986]: Received 36 observations.
[ INFO] [1489078594.202242961]: Occlusion manager returned 0 tracks
nnt_node: /usr/include/boost/smart_ptr/shared_ptr.hpp:648: typename boost::detail::sp_member_access<T>::type boost::shared_ptr<T>::operator->() const [with T = const spencer_tracking_msgs::DetectedPersons_<std::allocator<void> >; typename boost::detail::sp_member_access<T>::type = const spencer_tracking_msgs::DetectedPersons_<std::allocator<void> >*]: Assertion `px != 0' failed.

Thread 1 "nnt_node" received signal SIGABRT, Aborted.
0x00007ffff5b4d428 in __GI_raise (sig=sig@entry=6)
    at ../sysdeps/unix/sysv/linux/raise.c:54
54      ../sysdeps/unix/sysv/linux/raise.c: No such file or directory.
```

and the backtrace:

```
Reading symbols from /home/kota/spencer_kinetic_ws/devel/lib/srl_nearest_neighbor_tracker/nnt_node...done.
Starting program: /home/kota/spencer_kinetic_ws/devel/lib/srl_nearest_neighbor_tracker/nnt_node __name:=srl_nearest_neighbor_tracker __log:=/home/kota/.ros/log/f0bc9330-04e7-11e7-bd4e-64006a58bea8/spencer-perception_internal-people_tracking-srl_nearest_neighbor_tracker-55.log
[... 61 lines cut ...]
    () from /opt/ros/kinetic/lib/libroscpp.so
#20 0x00007ffff732672b in ros::spin() () from /opt/ros/kinetic/lib/libroscpp.so
#21 0x0000000000b12a25 in srl_nnt::ROSInterface::spin (this=0x7fffffffd7b0)
    at /home/kota/spencer_kinetic_ws/src/spencer_people_tracking/tracking/people/srl_nearest_neighbor_tracker/src/srl_nearest_neighbor_tracker/ros/ros_interface.cpp:85
#22 0x0000000000a93d3c in main (argc=1, argv=0x7fffffffe0e8)
    at /home/kota/spencer_kinetic_ws/src/spencer_people_tracking/tracking/people/srl_nearest_neighbor_tracker/src/srl_nearest_neighbor_tracker/ros/nodes/nnt_node.cpp:21
```

Actually, looks like `m_currentLowConfidenceDetections` was being used while pointing to null. I don't know how this impacts the rest of the system but at least if I do a check for null and return it doesn't seem to crash, and based on my 20 seconds of testing also seems to work:

```
diff --git a/tracking/people/srl_nearest_neighbor_tracker/src/srl_nearest_neighbor_tracker/missed_observation_recovery/low_confidence_observations_recovery.cpp b/tracking/people/srl_nearest_neighbor_tracker/src/srl_nearest_neighbor_tracker/missed_observation_recovery/low_confidence_observations_recovery.cpp
index af3adbe..300619f 100644
--- a/tracking/people/srl_nearest_neighbor_tracker/src/srl_nearest_neighbor_tracker/missed_observation_recovery/low_confidence_observations_recovery.cpp
[... cut ...] ed_observation_recovery/low_confidence_observations_recovery.cpp
@@ -78,6 +78,8 @@ void LowConfidenceObservationsRecovery::recoverObservations(const double current
 {
     // Check if normal and low-confidence detections are sufficiently synchronized
     m_numCyclesTotal++;
+
+    if (!m_currentLowConfidenceDetections) return;
     double timestampDelta = currentTime - m_currentLowConfidenceDetections->header.stamp.toSec();
     if(std::abs(timestampDelta) > m_maxTimestampDifference)
     {
```

tlind, 2017-03-09:
Yes, makes sense. What you see is a race condition when the regular detections arrive before the low-confidence detections, or if there are no low-confidence detections at all -- I guess we never tested this after refactoring the functionality into a separate class. Your fix is fine, can you open a PR for it?

femtogram, 2017-03-10:
Absolutely. Opened a PR.

By the way, this is not really the place to ask but i'm not sure where else would be better. I made some changes locally to rwth_upper_body_detector that use C++11 smart pointers (there was a segfault there as well, and the general consensus seems to be to avoid raw pointers). I don't think I saw any C++11/14 anywhere in spencer, so I figure I shouldn't do a pull request? I can switch it to using boost smart pointers instead. To make it worse these are on top of https://github.com/spencer-project/spencer_people_tracking/pull/14 which was closed due to not having the changes be isolated, so I suppose I would need to make PRs to fix some other things that rentt addresses as well.

closed 2017-03-10 by tlind

tlind, 2017-03-13:
@femtogram I just read your comment. For me, C++11 would be fine as long as the package still compiles under Ubuntu 14.04 / ROS Indigo.

However, I have a hard time following all the changes done by rentt in #14 by looking at the commits. Especially given that some commit messages appear twice (not sure if he was undoing and redoing some changes, or what happened). Looking at the [changed files](https://github.com/spencer-project/spencer_people_tracking/pull/14/files) however, it looks like not too many things have been changed. Maybe we can redo these changes in a somewhat more structured manner, without all the whitespace reformatting.

femtogram, 2017-03-14:
@tlind sounds great. I don't have access to the full sensor set right now (or any bag files with all the data), so I'm not in a great place to work on this today, but I'll start going through and make the appropriate changes over the next few days.

femtogram, 2017-03-14:
Actually, @tlind, my changes replaces most of those made by rentt. I'll look through to see if there are any relevant pieces that my patch doesn't cover. I just downloaded g++-4.8 (which I believe is the version in ubuntu 14.04) and made sure that everything I use is supported.

femtogram, 2017-03-20:
@tlind I'm having trouble recreating the segfaults related to the pointers I was having earlier with the latest build. I've submitted a small patch fixing an issue with indexing for the matrix, but haven't pushed the other changes since I can't test that.

commit 196847b5 (2017-03-10): Merge PR #22, fix for segfault in NNT without low-confidence detections, from femtogram/patch-1

Fix #19, segfault in nnt
commit 2232dd80 (2017-03-10): Check for shared_ptr being defined in nnt

Fixes #19