spencer-project/spencer_people_tracking #21: Try to compile package on ubunto 16.04 ROS kinetic

miguelmartins51, 2017-03-08:
HI guys.

  I am a beginner with ros and i found this project super interesting. I really want to try it with the kinect v1. Has anyone tried to compile this project on ubunto 16.04 wi
[error.docx](https://github.com/spencer-project/spencer_people_tracking/files/827473/error.docx)

th ros Kinetic ?

I am having this issues when i make may catkin_make. Anyone knows how to solve it ? 

fatal error: pluginlib/class_list_macros.h: No such file or directory
....
fatal error: nodelet/nodelet.h: No such file or directory

femtogram, 2017-03-08:
Please see the kinect-dev branch in joao-avelino's fork: https://github.com/joao-avelino/spencer_people_tracking/tree/kinetic_dev

Also, https://github.com/spencer-project/spencer_people_tracking/issues/19 and https://github.com/spencer-project/spencer_people_tracking/issues/20 should be of interest to you.

tlind, 2017-03-08:
Thanks for mentioning joao-avelino's fork, it contains several fixes to make the framework work on 16.04 and ROS Kinetic. Most issues arise from the fact that ROS Kinetic per default ships with OpenCV3, which had some breaking API changes. There were also some package dependencies missing and include directories not being referenced. Not sure why those issues didn't appear in 14.04.

I'll see if we can merge joao-avelino's changes back into this repo at some point, if he agrees to that, and everything still works under 14.04/Indigo.

femtogram, 2017-03-09:
@tlind well, seems like most of joao-avelino's changes were around rviz's switch to Qt5 from Qt4 (which would break compatibility with 14.04 and Indigo, though joao-avelino's patch seems to account for that), and to address the [issues with moc's C++ and preprocessor macro parsing](http://stackoverflow.com/questions/15455178/qt4-cgal-parse-error-at-boost-join), though he does indeed force building with OpenCV2 instead of 3 :)
Looks like his changes aren't under a different license, so it should be safe to just merge his branch back in even without his explicit permission. GitHub allows you to do a pull request even if you aren't the owner of the fork:
[image]

ROS versions really are a pain aren't they XD I really wish the docker instances had better networking support... Or better yet if ROS supported more than just the single OS and version. It would really simplify things.

tlind, 2017-03-13:
I contacted the author and he agreed for the changes to be merged back into our repository. I have subsequently merged PR #23 into master. I tested it under Ubuntu 14.04 / ROS Indigo and tracking_single_rgbd_sensor.launch works fine. Next, need to do further testing under 16.04 / ROS Kinetic as well as with the full SPENCER sensor setup.

Also still need to add an AUTHORS file and mention Joao for his great contribution.

tlind, 2017-03-16:
Status update: Full sensor setup confirmed to be working under 14.04. Not yet tested under 16.04.

miguelcrf450, 2017-03-21:
HI,

Thank you guys for your help. Like someone said before "ROS versions really are a pain". I tested with 14.04 and ros indigo and everything seems working well. I followed the tips from @femtogram and used the freenect with kinect v1. Now i will see how this work with 16.06 and kinetic.

gregoirefra, 2017-03-22:
For information, I tried it out on 16.04 with ROS Kinetic and there is a fail on build because of the hardcoded openCV 2. I'm trying to replace it with OpenCV 3 now but it seem's to be a really big task for the little knowledge I have on the difference between the two.

femtogram, 2017-03-22:
@gregoirefra OpenCV 3 is not backwards compatible with OpenCV 2. Honestly, you'll be better off just installing OpenCV 2. If I recall the `CMakeLists.txt` specifies a `find_package(OpenCV 2)` so you'll be best off just installing OpenCV 2 systemwide. You can install via `sudo apt install libopencv-dev`. I think a port to 3 doesn't make sense at this point since ROS Jade and Ubuntu 14.04 don't support OpenCV 3.

tlind, 2017-03-22:
Which package is failing and what's the error message?

I fully agree with @femtogram, porting the code to OpenCV 3 doesn't make sense. It should be no problem to install OpenCV2 (via apt-get) in parallel to the OpenCV 3 installation that comes with ROS Kinetic, as the former resides in `/usr/lib` and the latter resides in `/opt/ros`.

tlind, 2017-03-22:
@femtogram Regarding the README, I would propose to add `libopencv-dev` that you suggested as a build dependency to `package.xml` (this is currently not the case for the laser detectors and PCL people detector),  and then advise people to use `rosdep` to install the dependencies as documented in #29. Yet, of course we can still mention the OpenCV 2 dependency explicitly in the text.

tlind, 2017-03-24:
Done in 4afe68c and tested under ROS Kinetic.

closed 2017-03-24 by tlind