spencer-project/spencer_people_tracking #29: cannot compile in kinect

zubair123khan, 2017-03-22:
hi folks, 

i am using ros kinect and i am not able to compile spencer_tracking .. it need hell amount of libraries which diesnt come with the installation,,
i am is there any way to install all the libraries which is in use with spencer_tracking once for all,,??

please help

tlind, 2017-03-22:
Hi,

you can try the following commands after cloning the repository into your ROS workspace's `src` folder:

    rosdep update
    rosdep install --simulate -r --from-paths PATH_TO_YOUR_SRC_FOLDER --ignore-src

If everything looks good, remove the `--simulate` to actually execute it. Note I couldn't test this as I already have all dependencies installed on my system.

Before doing so please `git pull` once, as I just fixed a small bug related to `rosdep` in the master branch.

tlind, 2017-03-24:
I have added a few missing dependencies to various `package.xml` files required for compilation to succeed under ROS Kinetic. The above instructions using `rosdep install` should now do the trick.

closed 2017-03-24 by tlind