spencer-project/spencer_people_tracking #30: Problem running  tracking and visualization with a single RGB-D sensor

dmkarage, 2017-04-12:
Hi guys,

I compiled the spencer_people_tracking
 package (without HOG library) and I want to test the single RGB-D sensor tracking.I have an ASUS Pro sensor. When I launch the file I cannot see any point clouds in Rviz and the Upper Body and View Frustum displays are not valid showing "rgbd_front_top_rgb_optical_frame does not exist". Moreover in the terminal I get the following warning:

Failed to lookup transform from frame rgbd_front_top_rgb_optical_frame into target frame base_footprint! Dropping input DetectedPersons message on topic /spencer/perception_internal/detected_persons/rgbd_front_top/upper_body. Reason: "rgbd_front_top_rgb_optical_frame" passed to lookupTransform argument source_frame does not exist.

Could you help me to find out what I miss? Cause I think everything is ok with my sensor but somehow the frames of the sensor cannot be located. 

Thank you in advance

tlind, 2017-04-26:
Sorry for the late reply. I suggest running `rosrun tf tf_monitor` or `rosrun tf view_frames` to see which TF frames are being published at which intervals.

dmkarage, 2017-04-26:
No worries, I solved the issue by setting "publish_tf" = true in my Openn2 launch file. However I have a problem with the messages not being built. I will open a new thread.

Thank you