spencer-project/spencer_people_tracking #32: UpperBodyDetector not publishing anything

brinij, 2017-04-26:
Hello,

using Intel RealSense R200 RGB-D camera, I am launching tracking_single_rgbd_sensor.launch.
I've done remapping of "camera" namespace into "/spencer/sensors/rgbd_front_top", so I can see PointCloud data in RViz, and TF-s are also matching fine. Camera is mounted on height of 1.55 m.
Only problem that appear is NO DETECTION, meaning nothing is published on topics created by node : 
"/spencer/perception_internal/people_detection/rgbd_front_top/upper_body_detector"

My rosnode info /spencer/perception_internal/people_detection/rgbd_front_top/upper_body_detector : 

Node [/spencer/perception_internal/people_detection/rgbd_front_top/upper_body_detector]
Publications: 
 * /spencer/perception_internal/people_detection/rgbd_front_top/upper_body_detector/image/compressed/parameter_updates [dynamic_reconfigure/Config]
 * /spencer/perception_internal/people_detection/rgbd_front_top/upper_body_detector/image/compressed/parameter_descriptions [dynamic_reconfigure/ConfigDescription]
 * /spencer/perception_internal/people_detection/rgbd_front_top/upper_body_detector/image/compressedDepth/parameter_descriptions [dynamic_reconfigure/ConfigDescription]
 * /spencer/perception_internal/people_detection/rgbd_front_top/upper_body_detector/detections [rwth_perception_people_msgs/UpperBodyDetector]
 * /spencer/perception_internal/people_detection/rgbd_front_top/upper_body_detector/image/compressedDepth/parameter_updates [dynamic_reconfigure/Config]
 * /spencer/perception_internal/people_detection/rgbd_front_top/upper_body_detector/image [sensor_msgs/Image]
 * /spencer/perception_internal/detected_persons/rgbd_front_top/upper_body [spencer_tracking_msgs/DetectedPersons]
 * /diagnostics [diagnostic_msgs/DiagnosticArray]
 * /spencer/perception_internal/people_detection/rgbd_front_top/upper_body_detector/image/compressedDepth [sensor_msgs/CompressedImage]
 * /rosout [rosgraph_msgs/Log]
 * /spencer/perception_internal/people_detection/rgbd_front_top/upper_body_detector/image/theora/parameter_updates [dynamic_reconfigure/Config]
 * /spencer/perception_internal/people_detection/rgbd_front_top/upper_body_detector/bounding_box_centres [geometry_msgs/PoseArray]
 * /spencer/perception_internal/people_detection/rgbd_front_top/upper_body_detector/image/theora [theora_image_transport/Packet]
 * /spencer/perception_internal/people_detection/rgbd_front_top/upper_body_detector/image/theora/parameter_descriptions [dynamic_reconfigure/ConfigDescription]
 * /spencer/perception_internal/people_detection/rgbd_front_top/upper_body_detector/image/compressed [sensor_msgs/CompressedImage]

Subscriptions: 
 * /spencer/sensors/rgbd_front_top/depth/camera_info [sensor_msgs/CameraInfo]
 * /spencer/sensors/rgbd_front_top/ground_plane [rwth_perception_people_msgs/GroundPlane]
 * /spencer/sensors/rgbd_front_top/depth/image_rect [sensor_msgs/Image]
 * /spencer/sensors/rgbd_front_top/rgb/image_rect_color [sensor_msgs/Image]

Services: 
 * /spencer/perception_internal/people_detection/rgbd_front_top/upper_body_detector/get_loggers
 * /spencer/perception_internal/people_detection/rgbd_front_top/upper_body_detector/image/compressed/set_parameters
 * /spencer/perception_internal/people_detection/rgbd_front_top/upper_body_detector/set_logger_level
 * /spencer/perception_internal/people_detection/rgbd_front_top/upper_body_detector/image/compressedDepth/set_parameters
 * /spencer/perception_internal/people_detection/rgbd_front_top/upper_body_detector/image/theora/set_parameters

contacting node ..
Pid: 6775
Connections:
 * topic: /rosout
    * to: /rosout
    * direction: outbound
    * transport: TCPROS
 * topic: /spencer/perception_internal/people_detection/rgbd_front_top/upper_body_detector/image
    * to: /tracking_visualization_rviz
    * direction: outbound
    * transport: TCPROS
 * topic: /spencer/perception_internal/detected_persons/rgbd_front_top/upper_body
    * to: /spencer/perception_internal/detected_person_association/detected_person_association_nodelet_manager
    * direction: outbound
    * transport: TCPROS
 * topic: /spencer/perception_internal/detected_persons/rgbd_front_top/upper_body
    * to: /tracking_visualization_rviz
    * direction: outbound
    * transport: TCPROS
 * topic: /spencer/perception_internal/detected_persons/rgbd_front_top/upper_body
    * to: /rostopic_9449_1493195502422
    * direction: outbound
    * transport: TCPROS
 * topic: /spencer/sensors/rgbd_front_top/rgb/image_rect_color
    * to: /spencer/sensors/rgbd_front_top/rgbd_front_top_nodelet_manager (http:// MY ROS IP /)
    * direction: inbound
    * transport: TCPROS
 * topic: /spencer/sensors/rgbd_front_top/depth/camera_info
    * to: /spencer/sensors/rgbd_front_top/rgbd_front_top_nodelet_manager (http:// MY ROS IP /)
    * direction: inbound
    * transport: TCPROS
 * topic: /spencer/sensors/rgbd_front_top/ground_plane
    * to: /spencer/perception_internal/people_detection/ground_plane_front_top (http:// MY ROS IP/)
    * direction: inbound
    * transport: TCPROS
 * topic: /spencer/sensors/rgbd_front_top/depth/image_rect
    * to: /spencer/sensors/rgbd_front_top/rgbd_front_top_nodelet_manager (http:// MY ROS IP/)
    * direction: inbound
    * transport: TCPROS

I also checked "rostopic echo /spencer/perception/detected_persons" and there are also no messages. And all 4 subscribed topics of that node are working fine.

So, my question is : Is it problem in parameters of detector (or camera params in config file), so nothing is published because there is no human detection, or something else is wrong? What does detector do when there are no humans in field of view, does it stop publishing?

Thank you in advance!

tlind, 2017-04-26:
Hi,

when there are no humans in the field of view, the detector will continue publishing empty DetectedPersons messages on the `/spencer/perception_internal/detected_persons/rgbd_front_top/upper_body` topic, which then gets fused into `/spencer/perception/detected_persons`. So if there are no messages being published on the former, something else must be wrong.

I don't know the Intel RealSense camera, at which resolution is it publishing depth images? I believe the upper-body detector requires the resolution to be fixed at 640x480(?). That's also why it does not work with the Kinect v2 (see issue #4).

brinij, 2017-04-26:
Thanks for a quick reply!
Intel RealSense camera has rgb image resolution of 640x480, but depth image resolution of 480x360. 
Would it be possible to make it work by only changing the parameters dImWidth and dImHeight in config_Asus.inp?

tlind, 2017-05-07:
I think this is not sufficient; maybe scaling the depth image could help. See the recent comments in #4.

Charles-kang, 2017-08-08:
@brinij , Hi
I have encountered the same problem using MS Kinect V1, have you address this issue?

jcgarciaca, 2018-10-18:
@brinij 
I am also trying to run spencer example with a realsense camera. I set the namespace to `/spencer/sensors/rgbd_front_top`, but when running `roslaunch spencer_people_tracking_launch tracking_single_rgbd_sensor.launch height_above_ground:=1.6 load_driver:=false` it does not work. I suppose it is due to topics mismatch, below is the `rostopic list` command output. How should I remap the entire topics to match which spencer uses?

`/spencer/sensors/rgbd_front_top/aligned_depth_to_color/camera_info
/spencer/sensors/rgbd_front_top/aligned_depth_to_color/image_raw
/spencer/sensors/rgbd_front_top/aligned_depth_to_color/image_raw/compressed
/spencer/sensors/rgbd_front_top/aligned_depth_to_color/image_raw/compressed/parameter_descriptions
/spencer/sensors/rgbd_front_top/aligned_depth_to_color/image_raw/compressed/parameter_updates
/spencer/sensors/rgbd_front_top/aligned_depth_to_color/image_raw/compressedDepth
/spencer/sensors/rgbd_front_top/aligned_depth_to_color/image_raw/compressedDepth/parameter_descriptions
/spencer/sensors/rgbd_front_top/aligned_depth_to_color/image_raw/compressedDepth/parameter_updates
/spencer/sensors/rgbd_front_top/aligned_depth_to_color/image_raw/theora
/spencer/sensors/rgbd_front_top/aligned_depth_to_color/image_raw/theora/parameter_descriptions
/spencer/sensors/rgbd_front_top/aligned_depth_to_color/image_raw/theora/parameter_updates
/spencer/sensors/rgbd_front_top/aligned_depth_to_infra1/camera_info
/spencer/sensors/rgbd_front_top/aligned_depth_to_infra1/image_raw
/spencer/sensors/rgbd_front_top/aligned_depth_to_infra1/image_raw/compressed
/spencer/sensors/rgbd_front_top/aligned_depth_to_infra1/image_raw/compressed/parameter_descriptions
/spencer/sensors/rgbd_front_top/aligned_depth_to_infra1/image_raw/compressed/parameter_updates
/spencer/sensors/rgbd_front_top/aligned_depth_to_infra1/image_raw/compressedDepth
/spencer/sensors/rgbd_front_top/aligned_depth_to_infra1/image_raw/compressedDepth/parameter_descriptions
/spencer/sensors/rgbd_front_top/aligned_depth_to_infra1/image_raw/compressedDepth/parameter_updates
/spencer/sensors/rgbd_front_top/aligned_depth_to_infra1/image_raw/theora
/spencer/sensors/rgbd_front_top/aligned_depth_to_infra1/image_raw/theora/parameter_descriptions
/spencer/sensors/rgbd_front_top/aligned_depth_to_infra1/image_raw/theora/parameter_updates
/spencer/sensors/rgbd_front_top/color/camera_info
/spencer/sensors/rgbd_front_top/color/image_raw
/spencer/sensors/rgbd_front_top/color/image_raw/compressed
/spencer/sensors/rgbd_front_top/color/image_raw/compressed/parameter_descriptions
/spencer/sensors/rgbd_front_top/color/image_raw/compressed/parameter_updates
/spencer/sensors/rgbd_front_top/color/image_raw/compressedDepth
/spencer/sensors/rgbd_front_top/color/image_raw/compressedDepth/parameter_descriptions
/spencer/sensors/rgbd_front_top/color/image_raw/compressedDepth/parameter_updates
/spencer/sensors/rgbd_front_top/color/image_raw/theora
/spencer/sensors/rgbd_front_top/color/image_raw/theora/parameter_descriptions
/spencer/sensors/rgbd_front_top/color/image_raw/theora/parameter_updates
/spencer/sensors/rgbd_front_top/color/image_rect_color
/spencer/sensors/rgbd_front_top/color/image_rect_color/compressed
/spencer/sensors/rgbd_front_top/color/image_rect_color/compressed/parameter_descriptions
/spencer/sensors/rgbd_front_top/color/image_rect_color/compressed/parameter_updates
/spencer/sensors/rgbd_front_top/color/image_rect_color/compressedDepth
/spencer/sensors/rgbd_front_top/color/image_rect_color/compressedDepth/parameter_descriptions
/spencer/sensors/rgbd_front_top/color/image_rect_color/compressedDepth/parameter_updates
/spencer/sensors/rgbd_front_top/color/image_rect_color/theora
/spencer/sensors/rgbd_front_top/color/image_rect_color/theora/parameter_descriptions
/spencer/sensors/rgbd_front_top/color/image_rect_color/theora/parameter_updates
/spencer/sensors/rgbd_front_top/color_rectify_color/parameter_descriptions
/spencer/sensors/rgbd_front_top/color_rectify_color/parameter_updates
/spencer/sensors/rgbd_front_top/depth/camera_info
/spencer/sensors/rgbd_front_top/depth/image_rect_raw
/spencer/sensors/rgbd_front_top/depth/image_rect_raw/compressed
/spencer/sensors/rgbd_front_top/depth/image_rect_raw/compressed/parameter_descriptions
/spencer/sensors/rgbd_front_top/depth/image_rect_raw/compressed/parameter_updates
/spencer/sensors/rgbd_front_top/depth/image_rect_raw/compressedDepth
/spencer/sensors/rgbd_front_top/depth/image_rect_raw/compressedDepth/parameter_descriptions
/spencer/sensors/rgbd_front_top/depth/image_rect_raw/compressedDepth/parameter_updates
/spencer/sensors/rgbd_front_top/depth/image_rect_raw/theora
/spencer/sensors/rgbd_front_top/depth/image_rect_raw/theora/parameter_descriptions
/spencer/sensors/rgbd_front_top/depth/image_rect_raw/theora/parameter_updates
/spencer/sensors/rgbd_front_top/depth_registered/points
/spencer/sensors/rgbd_front_top/extrinsics/depth_to_color
/spencer/sensors/rgbd_front_top/extrinsics/depth_to_infra1
/spencer/sensors/rgbd_front_top/infra1/camera_info
/spencer/sensors/rgbd_front_top/infra1/image_rect_raw
/spencer/sensors/rgbd_front_top/infra1/image_rect_raw/compressed
/spencer/sensors/rgbd_front_top/infra1/image_rect_raw/compressed/parameter_descriptions
/spencer/sensors/rgbd_front_top/infra1/image_rect_raw/compressed/parameter_updates
/spencer/sensors/rgbd_front_top/infra1/image_rect_raw/compressedDepth
/spencer/sensors/rgbd_front_top/infra1/image_rect_raw/compressedDepth/parameter_descriptions
[... cut ...] infra1/image_rect_raw/compressedDepth/parameter_updates
/spencer/sensors/rgbd_front_top/infra1/image_rect_raw/theora
/spencer/sensors/rgbd_front_top/infra1/image_rect_raw/theora/parameter_descriptions
/spencer/sensors/rgbd_front_top/infra1/image_rect_raw/theora/parameter_updates
/spencer/sensors/rgbd_front_top/realsense2_camera_manager/bond
/spencer/sensors/rgbd_front_top/realsense2_camera_manager/parameter_descriptions
/spencer/sensors/rgbd_front_top/realsense2_camera_manager/parameter_updates`

dimaw24, 2019-02-06:
I'm having the same problem. Has anyone found a solution yet?

kivrakh, 2019-07-01:
The problem can be solved by changing the **relevant line** hardcoded in `base_realsense_node.cpp`

`_stream_name[RS2_STREAM_COLOR] = "color";`

This answer is referenced to:
> http://answers.ros.org/question/306463/is-it-possible-to-remap-cameracolor-topics-in-realsense2_camera/

kivrakh, 2019-07-01:
> The problem can be solved by changing the **relevant line** hardcoded in `base_realsense_node.cpp`
> 
> `_stream_name[RS2_STREAM_COLOR] = "color";`
> 
> This answer is referenced to:
> 
> > http://answers.ros.org/question/306463/is-it-possible-to-remap-cameracolor-topics-in-realsense2_camera/

Instead, you can directly use remap statement by putting just before the node description in the launch file