spencer-project/spencer_people_tracking #33: Messages not being built.

dmkarage, 2017-04-26:
Hi,

after successfully launching the tracking_single_rgbd_sensor.launch, I try to echo the /spencer/perception/tracked_persons topic. I receive the following error:

ERROR: Cannot load message class for [spencer_tracking_msgs/TrackedPersons]. Are your messages built? 

I know that this has to do either with CMakeLists.txt or the way that the package is built. I built the package as you recommend in the instructions.  The CMakeLists.txt in the spencer_people_tracking/launch/spencer_people_tracking_launch looks like: 

cmake_minimum_required(VERSION 2.8.3)
project(spencer_people_tracking_launch)

find_package(catkin REQUIRED COMPONENTS)

The CMakeLists.txt in spencer_people_tracking/messages/spencer_tracking_msgs/
looks like:

cmake_minimum_required(VERSION 2.8.3)
project(spencer_tracking_msgs)

find_package(catkin REQUIRED COMPONENTS
  roscpp
  roslib
  std_msgs
  geometry_msgs
  message_generation
)

################################################
## Declare ROS messages, services and actions ##
################################################

# Generate messages in the 'msg' folder
add_message_files(
   FILES
   DetectedPerson.msg
   DetectedPersons.msg
   CompositeDetectedPerson.msg
   CompositeDetectedPersons.msg
   TrackedPerson.msg
   TrackedPersons.msg
   TrackedPerson2d.msg
   TrackedPersons2d.msg
   TrackedGroup.msg
   TrackedGroups.msg
   ImmDebugInfo.msg
   ImmDebugInfos.msg
   TrackingTimingMetrics.msg
)

# Generate services in the 'srv' folder
add_service_files(
  FILES
  GetPersonTrajectories.srv
)

## Generate added messages and services with any dependencies listed here
generate_messages(
   DEPENDENCIES
   geometry_msgs
   std_msgs
)

###################################
## catkin specific configuration ##
###################################
catkin_package(
  CATKIN_DEPENDS roscpp std_msgs geometry_msgs message_runtime 
)

I did not change anything after building the package. My messages are not built even when I try to launch the tracking_on_robot.

What could be possibly wrong?

Thank you

tlind, 2017-04-26:
Maybe you are just not source'ing the workspace properly, before running roslaunch or rostopic echo?

dmkarage, 2017-04-26:
After building I did "source devel/setup.bash" in my ~/workspaces/spencer_ws. I do not know if anything else is needed. Also when I try roscd in the spenser packages it works, which means that it is probably correctly sourced. In the rostopic list I can see the several topics, but when I do echo, i receive the missing msg error.

closed 2017-05-01 by dmkarage

miguelcrf450, 2017-05-01:
dmkarage i have the same problem. Did you already resolved it? when i try echo the /spencer/perception/tracked_persons topic appears ERROR: Cannot load message class for [spencer_tracking_msgs/TrackedPersons]. Are your messages built?

dmkarage, 2017-05-02:
miguelcrf450 built it as suggested in the readme by the guys (with catkin build) and then try to source devel/setup.bash in ALL the terminals that you use. Hope it works!