spencer-project/spencer_people_tracking #34: Associating original detections from a track

ruffsl, 2017-05-03:
I'd like to be able to associate original detections from a track in order to then process those observations for classification purposes, e.g. recognise a tracked individual by appearance and assign/update the tracking ID to label or identity. In the [README.md#L60](https://github.com/spencer-project/spencer_people_tracking/blame/02db9e23002f405b9960c7b98bfeba5a86dee52d/README.md#L60), something quite equivalent to affect is aluded to:

> In case of detection-to-track fusion (currently not implemented), it is still advisable to publish a [CompositeDetectedPerson](/messages/spencer_tracking_msgs/msg/CompositeDetectedPerson.msg) message (via [CompositeDetectedPersons](/messages/spencer_tracking_msgs/msg/CompositeDetectedPersons.msg)) for each set of detections associated with a track, such that later on it is possible to go back to the original detections from a track, and lookup associated image bounding boxes etc. via the associated detection_id.

Currently, the [`spencer_tracking_msgs/TrackedPerson`](https://github.com/spencer-project/spencer_people_tracking/blob/34b3da20b2f082353f95214943cbec58826d47da/messages/spencer_tracking_msgs/msg/TrackedPerson.msg#L7) message type only includes one `detection_id` field. From what I traced so far, the `detection_id` is derived from the last observation in the array of observations used during its initialisation, or updated using only the observation from the  best pairing:

1. [`newTrack->observation = observations.back();`](https://github.com/spencer-project/spencer_people_tracking/blob/34b3da20b2f082353f95214943cbec58826d47da/tracking/people/srl_nearest_neighbor_tracker/src/srl_nearest_neighbor_tracker/nearest_neighbor_tracker.cpp#L326)
    * while initialising new tracks from candidates
2. ['track->observation = bestPairing->observation;'](https://github.com/spencer-project/spencer_people_tracking/blob/34b3da20b2f082353f95214943cbec58826d47da/tracking/people/srl_nearest_neighbor_tracker/src/srl_nearest_neighbor_tracker/nearest_neighbor_tracker.cpp#L234)
    * update track's observation from the best pairing
2. [`trackedPerson.detection_id = track->observation->id;`](https://github.com/spencer-project/spencer_people_tracking/blob/34b3da20b2f082353f95214943cbec58826d47da/tracking/people/srl_nearest_neighbor_tracker/src/srl_nearest_neighbor_tracker/ros/ros_interface.cpp#L155)
    * while publishing tracks

 Additionally, timing information from sensor input to processed output is not preserved, as detections with different modalities are fused, resulting in ill-defined timestamps to publish fused detections under. This also makes message capture/synchronization (say for post processing ROIs) using ROS topic filters via timing not as suitable. See simplified node graph for dataflow for [`tracking_on_bagfile.launch`](https://github.com/spencer-project/spencer_people_tracking/blob/879b7cc1cfe8b7dde0e97f7cbb6f0713717a9698/launch/spencer_people_tracking_launch/launch/tracking_on_bagfile.launch) (click to expand) 
[image]

My question is how could we determine to what track a given observation may have been paired with? Specifically, my goal is to monitor "unlabeled" tracks, find upper body ROI pairings with those given tracks, search and recognise faces from those corresponding ROIs, then "label" those track IDs to identities as their faces come into view.

Thanks for any guidance.

tlind, 2017-05-03:
Hi,

if you are using the standard detection-to-detection fusion pipeline, the detection_id in the TrackedPerson message will contain the ID of the _fused_ detection. To find out which original detections have been fused together, you have to subscribe to the `detected_persons_composite` topic, and keep a memory of those fused detections until the output from the tracker is available.

So instead of synchronizing via timestamp, you buffer the CompositeDetectedPersons until you are able to process them. As the assigned (composite) detection IDs are monotonically increasing, detection IDs from different frames won't collide with each other.

I once wrote a Python module for this purpose, but I haven't tested it in a long while so I don't know if it still works. But maybe this can serve as an inspiration:

https://github.com/spencer-project/spencer_people_tracking/blob/master/detection/spencer_detected_person_association/src/spencer_detected_person_association/__init__.py

There is a usage example for the TrackSynchronizer class starting from line 172:

```python
def callback(trackAssociation, humanAttributes):
    # Do something useful with the human attributes message...
    # Use trackAssociation.lookupTrackId(detectionId) to look up the track associated with a particular detection (may be None)

human_attribute_sub = message_filters.Subscriber('/spencer/perception_internal/human_attributes/rgbd_front_top', HumanAttributes)
track_sync = spencer_detected_person_association.TrackSynchronizer(human_attribute_sub, 100)
track_sync.registerCallback(callback)
```

closed 2017-05-07 by tlind

ruffsl, 2017-05-12:
Alright, so we've used the TrackSynchronizer and are encountering some steady state synchronization issues. Here is our current approach: [main.py](https://github.com/CogRob/spencer_people_tracking/blob/ba200449e1711906eb0dbf3757c1858e4b88ac1b/classification/face_classifier/openface_classifier/src/main.py)

We are using the [TrackSynchronizer](https://github.com/CogRob/spencer_people_tracking/blob/ba200449e1711906eb0dbf3757c1858e4b88ac1b/classification/face_classifier/openface_classifier/src/main.py#L57-L59) with the upper body detections then rebroadcasting the detection and track_id correspondence to a new message topic. In this file, we are using the [ApproximateTimeSynchronizer](https://github.com/CogRob/spencer_people_tracking/blob/ba200449e1711906eb0dbf3757c1858e4b88ac1b/classification/face_classifier/openface_classifier/src/main.py#L40-L55) to filter the RGB image, the upper body detections, the corresponding ROIs, as well as the track associations mentioned previously.

The current issues we are encountering is that the callback will proceed as expected, publishing image crops with matching track_id overlay, but only for limited time after startup; around 10 seconds. Thereafter the callback encounters issues where the list of raw detections and track associations share [no common detection ids](https://github.com/CogRob/spencer_people_tracking/blob/ba200449e1711906eb0dbf3757c1858e4b88ac1b/classification/face_classifier/openface_classifier/src/main.py#L89).

We hypothesize this maybe due to the fact that the phase shift in publication timestamps between the original detections message and the track association topic we create grows beyond the period of the common publication rate, as such that the ApproximateTimeFilter is aligning messages that do not truly correspond with each other.

Perhaps, you could take a look at our approach and recommend any modifications?

PR #35 (2017-05-07): Updating TrackSynchronizer
Updating `TrackSynchronizer` to use current message types and topics namespaces.
Fixes #34