spencer-project/spencer_people_tracking #39: NVIDIA Jetson TX2 Developer Kit

miguelcrf450, 2017-05-24:
Hi guys,

   I have an hardware question this time. I want to deploy this project on a small embedded system from Nvidia. I don´t have the board yet but i am thinking on buy it. It is called the "NVIDIA Jetson TX2 Developer Kit" and supports ubunto 16.04, 64 bits, includes ROS compatibility, and seems to be proper to computer vision.

Anyone that have worked with that board have any tip or sugestion? Do you think it is capable of run spencer with 2 laser scanner and one depth camera?

Here is the specs of the board:

`NVIDIA Jetson TX2   	

CPU              ARM Cortex-A57 (quad-core) @ 2GHz + NVIDIA Denver2 (dual-core) @ 2GHz
GPU              256-core Pascal  @ 1300MHz
Memory	     8GB 128-bit LPDDR4 @ 1866Mhz |  59.7 GB/s
Storage        32GB eMMC 5.1
Encoder*	     4Kp60, (3x) 4Kp30, (8x) 1080p30
Decoder*	     (2x) 4Kp60
Camera†	     12 lanes MIPI CSI-2 | 2.5 Gb/sec per lane | 1400 megapixels/sec ISP
Display	      2x HDMI 2.0 / DP 1.2 / eDP 1.2 | 2x MIPI DSI
Wireless         802.11a/b/g/n/ac 2×2 867Mbps | Bluetooth 4.1
Ethernet	      10/100/1000 BASE-T Ethernet
USB	               USB 3.0 + USB 2.0
PCIe	               Gen 2 | 1×4 + 1×1 or 2×1 + 1×2
CAN	               Dual CAN bus controller
Misc I/O	       UART, SPI, I2C, I2S, GPIOs
Socket	       400-pin Samtec board-to-board connector, 50x87mm`

Thanks,

cagbal, 2017-05-31:
I am also interested in deploying Spencer code on TX2. Please keep this topic up to date. TX2 should be capable of running it in real time, since currently I am running the tracker with just single RGBD camera on my laptop which has Intel® Core™ i7-6820HK CPU @ 2.70GHz × 8 but not GPU.

archenroot, 2017-06-18:
I am also interested not only on deployment, but also in performance. I might be interested in some extension implementation, etc. so seing one has still plenty of unused resources while running spencer as is could be valuable before obtaining the Nvidia kit...

miguelcrf450, 2017-06-19:
I din´t bought the board yet so i can´t give the feedback. I am trying to find someone that has one jetson to test.

miguelcrf450, 2017-06-22:
Hi guys,

   I have a question that maybe somebody can help me. In this moment i am only using the upper_body_detector and not using cuda. 
   My question is if the spencer framework requires much GPU? For me this framework is much more CPU intensive that GPU. And i am thinking in search for another board with a much better processor than use the jetson.

Thank guys

cagbal, 2017-06-22:
Hi, unfortunately, I am using it only with CPU.

tlind, 2017-06-22:
@miguelcrf450, if you are only using the upper_body_detector (but not groundHOG), the framework will not use any GPU at all if I recall correctly, unless you run Rviz etc. onboard for visualization. Therefore, you are correct that a board with a good CPU (esp. multiple cores/hyperthreading) will currently make more sense.

This might change in the next 1-3 years once more deep learning-based person detectors become available and somebody integrates them into the framework. The tracker itself rarely uses any CPU.