spencer-project/spencer_people_tracking #42: Packages depending on cv_bridge incorrectly linked against OpenCV3

tlind, 2017-06-09:
Since we are now starting to build packages on the LCAS buildfarm, an issue appears under ROS Kinetic / Ubuntu 14.04 that several packages which depend on `cv_bridge` are linked against OpenCV 3.1 on the buildfarm. The issue only becomes apparent once these packages are installed on the host system, and the C++ binaries are executed. Affected packages include:

* rwth_upper_body_detector
* video_to_bagfile

Even after installing `ros-kinetic-opencv3`, the error messages by the dynamic linker don't disappear, as the ROS OpenCV version is 3.2 (instead of 3.1).

Is there any workaround that does not involve having our own OpenCV 2 version of cv_bridge?

tlind, 2017-06-09:
The comment by Vincent Rabaud is relevant to us: https://github.com/ros-perception/vision_opencv/issues/143
_"If you link to other official ROS packages that use OpenCV (like cv_bridge), you need to use OpenCV3."_

Now, for `rwth_upper_body_detector` and `video_to_bagfile` switching to OpenCV 3 might not matter (because they e.g. do not use the ML part of OpenCV). However, then we could run into issues on ROS Indigo.

tlind, 2017-06-09:
This issue does not appear when building from source on ROS Kinetic. In that case, the relevant packages link against the installed OpenCV 3.2 (shipped with Kinetic):

    ldd video_to_bagfile:
      libcv_bridge.so => /opt/ros/kinetic/lib/libcv_bridge.so (0x00007f76d5a81000)
	  libopencv_core3.so.3.2 => /opt/ros/kinetic/lib/libopencv_core3.so.3.2 (0x00007f76d4db3000)
	  libopencv_imgcodecs3.so.3.2 => /opt/ros/kinetic/lib/libopencv_imgcodecs3.so.3.2 (0x00007f76d498e000)
	  libopencv_videoio3.so.3.2 => /opt/ros/kinetic/lib/libopencv_videoio3.so.3.2 (0x00007f76d4765000)

Not sure if this is a good solution in all cases, but at least for these packages, the change from OpenCV 2 to 3 did not break anything.

Does the buildfarm somehow still have an old version of cv_bridge? Maybe it needs its rosdistro to be updated?

tlind, 2017-06-12:
@marc-hanheide Any idea if this is buildfarm-related? Why is cv_bridge still linking against OpenCV 3.1 (instead of 3.2) on the buildfarm?

marc-hanheide, 2017-06-12:
Are you sure the released version is linked against 3.1? All evidence I have from the buildfarm suggests its linked against 3.2:
```
/usr/lib/ccache/x86_64-linux-gnu-g++   -g -O2 -fstack-protector-strong -Wformat -Werror=format-security -DNDEBUG -Wdate-time -D_FORTIFY_SOURCE=2    CMakeFiles/video_to_bagfile.dir/src/video_to_bagfile/video_to_bagfile.cpp.o  -o devel/lib/video_to_bagfile/video_to_bagfile -rdynamic /opt/ros/kinetic/lib/libimage_transport.so /opt/ros/kinetic/lib/libmessage_filters.so /opt/ros/kinetic/lib/libclass_loader.so -lPocoFoundation -ldl /opt/ros/kinetic/lib/libroslib.so /opt/ros/kinetic/lib/librospack.so -lpython2.7 -ltinyxml /opt/ros/kinetic/lib/librosbag.so /opt/ros/kinetic/lib/librosbag_storage.so -lboost_program_options /opt/ros/kinetic/lib/libroslz4.so -llz4 /opt/ros/kinetic/lib/libtopic_tools.so /opt/ros/kinetic/lib/libroscpp.so -lboost_signals -lboost_filesystem /opt/ros/kinetic/lib/libxmlrpcpp.so /opt/ros/kinetic/lib/libcv_bridge.so /opt/ros/kinetic/lib/libopencv_calib3d3.so.3.2.0 /opt/ros/kinetic/lib/libopencv_core3.so.3.2.0 /opt/ros/kinetic/lib/libopencv_features2d3.so.3.2.0 /opt/ros/kinetic/lib/libopencv_flann3.so.3.2.0 /opt/ros/kinetic/lib/libopencv_highgui3.so.3.2.0 /opt/ros/kinetic/lib/libopencv_imgcodecs3.so.3.2.0 /opt/ros/kinetic/lib/libopencv_imgproc3.so.3.2.0 /opt/ros/kinetic/lib/libopencv_ml3.so.3.2.0 /opt/ros/kinetic/lib/libopencv_objdetect3.so.3.2.0 /opt/ros/kinetic/lib/libopencv_photo3.so.3.2.0 /opt/ros/kinetic/lib/libopencv_shape3.so.3.2.0 /opt/ros/kinetic/lib/libopencv_stitching3.so.3.2.0 /opt/ros/kinetic/lib/libopencv_superres3.so.3.2.0 /opt/ros/kinetic/lib/libopencv_video3.so.3.2.0 /opt/ros/kinetic/lib/libopencv_videoio3.so.3.2.0 /opt/ros/kinetic/lib/libopencv_videostab3.so.3.2.0 /opt/ros/kinetic/lib/libopencv_viz3.so.3.2.0 /opt/ros/kinetic/lib/libopencv_aruco3.so.3.2.0 /opt/ros/kinetic/lib/libopencv_bgsegm3.so.3.2.0 /opt/ros/kinetic/lib/libopencv_bioinspired3.so.3.2.0 /opt/ros/kinetic/lib/libopencv_ccalib3.so.3.2.0 /opt/ros/kinetic/lib/libopencv_cvv3.so.3.2.0 /opt/ros/kinetic/lib/libopencv_datasets3.so.3.2.0 /opt/ros/kinetic/lib/libopencv_dpm3.so.3.2.0 /opt/ros/kinetic/lib/libopencv_face3.so.3.2.0 /opt/ros/kinetic/lib/libopencv_fuzzy3.so.3.2.0 /opt/ros/kinetic/lib/libopencv_hdf3.so.3.2.0 /opt/ros/kinetic/lib/libopencv_line_descriptor3.so.3.2.0 /opt/ros/kinetic/lib/libopencv_optflow3.so.3.2.0 /opt/ros/kinetic/lib/libopencv_phase_unwrapping3.so.3.2.0 /opt/ros/kinetic/lib/libopencv_plot3.so.3.2.0 /opt/ros/kinetic/lib/libopencv_reg3.so.3.2.0 /opt/ros/kinetic/lib/libopencv_rgbd3.so.3.2.0 /opt/ros/kinetic/lib/libopencv_saliency3.so.3.2.0 /opt/ros/kinetic/lib/libopencv_stereo3.so.3.2.0 /opt/ros/kinetic/lib/libopencv_structured_light3.so.3.2.0 /opt/ros/kinetic/lib/libopencv_surface_matching3.so.3.2.0 /opt/ros/kinetic/lib/libopencv_text3.so.3.2.0 /opt/ros/kinetic/lib/libopencv_xfeatures2d3.so.3.2.0 /opt/ros/kinetic/lib/libopencv_ximgproc3.so.3.2.0 /opt/ros/kinetic/lib/libopencv_xobjdetect3.so.3.2.0 /opt/ros/kinetic/lib/libopencv_xphoto3.so.3.2.0 /opt/ros/kinetic/lib/librosconsole.so /opt/ros/kinetic/lib/librosconsole_log4cxx.so /opt/ros/kinetic/lib/librosconsole_backend_interface.so -llog4cxx -lboost_regex /opt/ros/kinetic/lib/libroscpp_serialization.so /opt/ros/kinetic/lib/librostime.so /opt/ros/kinetic/lib/libcpp_common.so -lboost_system -lboost_thread -lboost_chrono -lboost_date_time -lboost_atomic -lpthread -lconsole_bridge /opt/ros/kinetic/lib/libopencv_highgui3.so.3.2.0 /opt/ros/kinetic/lib/libopencv_imgproc3.so.3.2.0 /opt/ros/kinetic/lib/libopencv_core3.so.3.2.0 -Wl,-rpath,/opt/ros/kinetic/lib: -Wl,-rpath-link,/opt/ros/kinetic/lib 
```

from https://lcas.lincoln.ac.uk/buildfarm/view/Kbin_uX64/job/Kbin_uX64__video_to_bagfile__ubuntu_xenial_amd64__binary/lastSuccessfulBuild/console

marc-hanheide, 2017-06-12:
I have just done a fresh Kinetic install from released packages and I get:
```
ldd /opt/ros/kinetic/lib/video_to_bagfile/video_to_bagfile  | grep opencv
        libopencv_core3.so.3.2 => /opt/ros/kinetic/lib/libopencv_core3.so.3.2 (0x00007f203c8b1000)
        libopencv_imgcodecs3.so.3.2 => /opt/ros/kinetic/lib/libopencv_imgcodecs3.so.3.2 (0x00007f203c48d000)
        libopencv_videoio3.so.3.2 => /opt/ros/kinetic/lib/libopencv_videoio3.so.3.2 (0x00007f203c264000)
        libopencv_imgproc3.so.3.2 => /opt/ros/kinetic/lib/libopencv_imgproc3.so.3.2 (0x00007f2038029000)
```

which looks good to me.

tlind, 2017-06-12:
Ok thanks, that's interesting. Need to check what's going on, maybe something's wrong with my Docker container

tlind, 2017-06-12:
Issue has disappeared after running `apt-get dist-upgrade`. So probably I had an old version of `cv_bridge` in my Docker container.

closed 2017-06-12 by tlind