spencer-project/spencer_people_tracking #45: OpenCV 3.x suppport

guchaojie, 2017-07-19:
Hi everyone, I have ROS Kinetic on Ubuntu 16.04. As stated in the documentation, it is supported by OpenCV 3.1, while all the previous ROS versions are supported by OpenCV 2.4.
There is any plan to support OpenCV 3.x for spencer_people_tracking.

JGMonroy, 2017-10-11:
Hi there,
I'm also facing the same problem as guchaojie, problems with OpenCV3.x when using these pkgs on Ubuntu 16.04 and ROS Kinetic. Any thoughts on moving to OpenCV3?

tlind, 2017-11-25:
We are not planning to migrate the framework to OpenCV3, as some components (e.g. the laser detectors) heavily rely on the OpenCV 2 internal  machine learningAPI.

However, it is possible to get it running on OpenCV3 without significant effort if you have installed OpenCV2 and OpenCV3 side-by-side. You essentially need to fork [vision-opencv](https://github.com/ros-perception/vision_opencv), rename e.g. the cv_bridge package to cv_bridge2, and then make the relevant SPENCER packages depend on cv_bridge2 instead of cv_bridge. Inside the `CMakeLists.txt` of the cv_bridge2 packages, use `find_package(OpenCV 2)` to enforce the old version of OpenCV being used.

bbrito, 2019-01-29:
Another solution is to install Opencv2 by source:

http://mohsaad.com/2017/07/17/Installing-ROS-With-OpenCV/