spencer-project/spencer_people_tracking #49: How to generate the annotated logfiles for laser: train_from_logfile.launch

lanyaye, 2017-11-21:
Hello, 
I have one question about the **train_from_logfile.launch**, it will use old 2D laserscan logfiles in CARMEN format as input. for example:

>     <arg name="filename" default="$(find srl_nearest_neighbor_tracker)/data/logfiles/fr_aspekt1.lab"/> 
>     <arg name="filename" default="$(find srl_nearest_neighbor_tracker)/data/logfiles/fr_mainstation1_cut.lab"/> -->
>     <arg name="filename" default="/home/linder/Datasets/kindercar_oct2013/rathausgasse.log"/>

I know that fr_aspekt1.lab and fr_mainstation1_cut.lab is coming from Annotated 2D laser datasets (Social Robotics Labs, University of Freiburg): [http://www2.informatik.uni-freiburg.de/~luber/people_tracker/logfiles/logfiles.html](url)

Then how do we training the laser data for different laser type and diferent height?

Are there any node which can publish the annotated data directly without this old logfile: `srl_laser_segmentation/LaserscanSegmentation` with the annotations (each laser endpoint is labelled with an integer label, corresponding to the track ID)
since the node **train_laser_detector** need this annotated data.

Thank you!

tlind, 2017-11-25:
As you probably saw, the package [srl_tracking_logfile_import](https://github.com/spencer-project/spencer_people_tracking/tree/master/tracking/people/srl_tracking_logfile_import) provides a node which reads old CARMEN logfiles, and publishes them as `LaserscanSegmentation` plus `LaserScan`. This is because we used an old annotation tool (from our research on multi-hypothesis tracking), where you would select individual laser points for annotation, which outputted annotated CARMEN logfiles. I don't have access to that tool anymore, though.

An alternative would be to use the [track annotation tool](https://github.com/spencer-project/spencer_people_tracking/blob/master/utils/track_annotation_tool/launch/gui.launch), where you essentially just annotate trajectory points. It publishes `spencer_tracking_msgs/TrackedPersons`, which you would then have to match against a laser scan segmentation using some fixed person radius.

In his ICRA'15 paper, Angus Leigh instead recommended the following approach:
* Generate negative training samples by driving around a robot in an empty room (no people around). Then you can just run the laser scan segmentation, and every laser endpoint can be marked as negative.
* Generate positive training samples by driving a robot around in a room full of people, but with no other objects or walls closeby. Then consider all resulting laser scan segments as positive samples.

smail1229, 2017-12-14:
Excuse me, are there any tips about running the track annotation tool?

It seems that simply running : `roslaunch track_annotation_tool gui.launch` can not get what I expected.
The node it create doesn't subscribe topic like /laser

Thank you.

closed 2017-12-29 by lanyaye

tlind, 2017-12-29:
@smail1229 Sorry I didn't see the notification about your comment earlier. The track annotation tool won't subscribe to any sensor-specific ROS topic, as it is a multi-modal 3D annotation tool that is not tied to a specific sensor. Instead, you annotate trajectory points directly in 3D space. However, the RViz instance that should be launched with the launch file you mentioned, can of course subscribe to any laser/point cloud topic and visualize it. You then use the toolbar buttons in Rviz to place trajectory waypoints at the appropriate locations.