spencer-project/spencer_people_tracking #51: Run example from a single .bagfile

iqedgarmg, 2018-03-27:
Hi everyone! 

I am trying to run the first tutorial using a single .bag file (in order to do this with my own RGB-D camera). For this purpose I did the following procedure:  

**1.- Merge the .bag files**
I merge 2 of the .bag files that are generated with the download_example_bagfiles.sh script (front_rgbd.bag and tf_odom.bag). This new .bag file called "Spencer.bag" has the following topics: 

- /spencer/sensors/rgbd_front_top/depth/camera_info
- /spencer/sensors/rgbd_front_top/depth/image_raw/compressedDepth
- /spencer/sensors/rgbd_front_top/rgb/camera_info
- /spencer/sensors/rgbd_front_top/rgb/image_raw
- /spencer/sensors/rgbd_front_top/rgb/image_raw/compressed
- /spencer/sensors/rgbd_front_top/rgb/image_rect_color/theora
- /tf

**2.- Modify launch file**
I uncomment and modify the following lines on tracking_on_bagfile.launch file: 

- `<arg name="bagfile_folder" default="$(env HOME)/spencer-people-tracking-ws/bags/"/>`
In order to especify my own bag directory. 

- `<arg name="files" value="$(arg bagfile_folder)/Spencer.bag"/>`
In order to indicate the merge .bag file.

- `<arg name="folder" value="$(arg bagfile_folder)"/>`
This line was commented, also I disable the front_laser and both rears_rgbd_laser args  in order to use only rgb and depth images.

But altough the RGB point cloud is showed on Rviz, no detection is processed. If I try with separate .bag files (front_rgbd.bag and tf_odom.bag) on my own folder using the "folder" arg the node works fine.

I really appreciate any help to solve this issue. 

Thanks in advance for your response

tlind, 2018-10-08:
Not sure why you are merging the bagfiles, as rosbag play can automatically do that during playback.
But nevertheless, it should work.

I would suggest to use rosnode list and rosnode info to go through the nodes (starting at the bottom with the srl_nearest_neighbor_tracker or nnt_node), to see if required topics are unconnected.

closed 2018-10-08 by tlind

kivrakh, 2019-04-02:
Hello,

I had the same issue with @iqedgarmg. To try the framework with my own bag data:

First of all, I carefully investigated the rosnode relationships and came up with the same topics as @iqedgarmg and recorded:

- /spencer/sensors/rgbd_front_top/depth/image_raw/compressedDepth
- /spencer/sensors/rgbd_front_top/rgb/image_raw/compressed
- /spencer/sensors/rgbd_front_top/rgb/camera_info
 - /spencer/sensors/rgbd_front_top/depth/camera_info
- /tf

Then use "image_transport" package to uncompress or republish:
- /spencer/sensors/rgbd_front_top/rgb/image_raw
- /spencer/sensors/rgbd_front_top/depth/image_raw

Finally use "openni2_launch" to publish:
- /spencer/sensors/rgbd_front_top/depth/image_rect
- /spencer/sensors/rgbd_front_top/depth/camera_info
- /spencer/sensors/rgbd_front_top/depth/image_rect_color

Although I see all the connections is done on rqt_graph and echo the topic. I cannot see any detection. This issue took my two days.

Thank you in advance.

iqedgarmg, 2019-04-02:
Hi @kivrakh, I remember that the problem was related to the stream codec of the depth, that needs to be the same that the example bag. You can do this with an OpenCV function, and check the correct codec with the "encoding" header of rosbag example provided by the package.. 

You can check this and if you need help I can recheck in my old files.

Greetings!

kivrakh, 2019-04-03:
Hi @iqedgarmg, I would be happy if you could check your solution and share with all of us. I think everyone who creates their own bag files encounters a similar problem!. I really appreciate that!

kivrakh, 2019-04-03:
When I run the following command
`roslaunch spencer_people_tracking_launch tracking_on_bagfile.launch`

The encodings of:
`rostopic echo /spencer/sensors/rgbd_front_top/depth/image_raw/encoding`
`rostopic echo /spencer/sensors/rgbd_front_top/rgb/image_raw/encoding`

are **16UC1** and **rgb8**, respectively. They are the same with my bag file outputs topic encodings..

kivrakh, 2019-04-04:
Finally, I got the point. The problem was related to **timestamps synchronization**. Because upper_body_detector uses message time synchronizer applied to the depth, camera_info and ground plane topics:

- `spencer/sensors/rgbd_front_top/depth/image_rect/`
- `spencer/sensors/rgbd_front_top/depth/camera_info/`
- `spencer/sensors/rgbd_front_top/ground_plane/`

That is, all these topics publish rate has to be the same. Ground_plane node rate was 30.0 by default. However, my publish rate of depth image and camera info nodes was about 4Hz. When I synchronize all these topics the problem was solved.

germal, 2020-11-30:
Hi @kivrakh ,
I am trying to adapt Spencer to d435 too.I have my topics syncronized with ground plane but the upper_body_detector still doesn't detect.
I modified topics name is modified as well. Could you advice what should I pay attention further ?
Thanks & Regards
germal

AlessandroMelino, 2021-03-09:
Hello @kivrakh and @germal , I am facing similar issues with a Realsense D435 and Upper Body Detector. Did you make it work?

I would ask 2 question:

1. How can I check if the topics are syncronized?
2. How is the correct way to set the ground plane?

Thanks in advance.
Alessandro

kivrakh, 2021-03-09:
Hello @AlessandroMelino,
Yes, we did make it work finally.

1. You should check the publish rate of that three topics by:

-  `rostopic hz spencer/sensors/rgbd_front_top/depth/image_rect/`
-   `rostopic hz spencer/sensors/rgbd_front_top/depth/camera_info/`
-   `rostopic hz spencer/sensors/rgbd_front_top/ground_plane/`

2. In `spencer_people_tracking_launch)/launch/detectors/front_rgbd_detectors.launch launch` file you can set the publish rate of ground plane using
- `<param name="rate" value="30.0"/>`

Hope this helps.

AlessandroMelino, 2021-03-10:
Hello @kivrakh, thank you for your answer.

I have checked the camera_info and ground_plane frequencies and they are ok, but the image_rect topic that exists but doesn't publish new messages. Instead I have a topic called /spencer/sensors/rgbd_front_top/depth/image_rect_raw/ which is the right topic the detector should subscribe. I tried remmapping it but didn't work (`<remap from="/spencer/sensors/rgbd_front_top/depth/image_rect_raw" to="/spencer/sensors/rgbd_front_top/depth/image_rect"/`>).

These are the topics published by the Realsense:

[image]

This is the topic I was talking about:

[image]

Am I doing it right?

Best regards.
Alessandro

germal, 2021-03-13:
Hello @AlessandroMelino , I've just seen your message, I don't use Spencer since some months but I remember I modified somewhere this frequency.I will getting back tomorrow after checking !

AlessandroMelino, 2021-03-15:
> Hello @AlessandroMelino , I've just seen your message, I don't use Spencer since some months but I remember I modified somewhere this frequency.I will getting back tomorrow after checking !

Okey, thank you for your answer. I am getting this frequencies if it is useful for you: 
[image]

I am getting too this warning when I launch the detector:
`[ WARN] [1615794471.342697164]: Failed transform lookup in rwth_ground_plane/tf_based_fixed_gp -- maybe the RGB-D drivers are not yet running!? Reason: "rgbd_front_top_depth_optical_frame" passed to lookupTransform argument target_frame does not exist. . Message will re-appear within 20 seconds.`

I don't know if this is important, but it doesn't appear any more.

Thanks in advance.
Alessandro

germal, 2021-03-15:
Hello Alessandro,

I couldn't get access to my code till now , but in the meantime I have found the following notes 

modify publishrate 30 -> 15 in the following files

- launch/spencer_people_tracking_launch/launch/detectors/front_rgbd_detectors.launch

-utils/rwth_ground_plane/src/tf_based_fixed_gp.cpp:    private_node_handle_.param("rate", publishRate, 30.0);
-utils/rwth_ground_plane/src/really_fixed_gp.cpp:    private_node_handle_.param("rate", publishRate, 30.0);

and recompile with catkin_make or catkin build

tweak also this parameter  height_above_ground in  roslaunch spencer_people_tracking_launch tracking_single_rgbd_sensor.launch 

Hope this help in the meantime ! Feel free to contact me on mail
Regards

AlessandroMelino, 2021-03-16:
Thank you for your answer @germal. 

So now I have done these changes and the publish rate of the ground_plane is 15 Hz:

[image]

But still, no detections.

I have set the height_above_ground parameter in 0.96 meters, as the height of the sensor respect to the ground. Must I try some other values?

By the way, I also set the tf of my sensor in that value. Can this cause any problem?

[image]

Thanks in advance for all the help.

Best regards.
Alessandro

germal, 2021-03-16:
Hey Alessandro,
In case you haven't done yet, you have to set depth scale to 1000 in the asus.yaml file.maybe this can resolve.
Regards !

kivrakh, 2021-03-16:
I advise you to check the depth and rgb image frequency.  There is a line 330 of code in `~/catkin_ws/src/spencer_people_tracking-melodic/detection/rgbd_detectors/rwth_upper_body_detector/src/main.cpp`. The difference in frequencies should not be much than 0.2.

```
// Creating a ros image with the detection results an publishing it
if(vis && color_image && abs((depth->header.stamp - color_image->header.stamp).toSec()) < 0.2) {  // only if color image is not outdated (not using a synchronizer!)
```

AlessandroMelino, 2021-03-17:
Hello @germal and @kivrakh. 

Finally it worked. It was the depth scale of the .yaml file, I have changed it to 1000 and it is detecting people.

Thanks a lot for all the help.

Best regards.
Alessandro

kivrakh, 2021-03-18:
I am happy that you find a solution to your problem. I have not experienced the depth scale parameter. Thank you for all.