spencer-project/spencer_people_tracking #58: can retrain?

gongyue666, 2018-08-14:
Can the detector be retrained by my own dataset? 
By the way there are too many nodes and .launch files, do they all needed?Seem too large for me

tlind, 2018-10-08:
Which detector?
The launch files and nodes are provided for maximum flexibility and deployment of different kinds of robots. This is a relatively generic framework for person tracking from (mobile) robots, not an application tailored to a specific robot.

CodeToPoem, 2019-02-24:
Hello,
I have a problem with the 'groups_probabilistic_small.model' in 'spencer_social_relations'. Can I see the origin datasets that used to retrain svm model?

tlind, 2019-02-24:
I think I might still have an annotated CARMEN logfile of the groundtruth tracks/groups. Give me some time to dig that out from an old computer. I cannot share the visual data (camera image stream) for privacy ressons.

CodeToPoem, 2019-02-25:
Thanks a lot. @tlind

tlind, 2019-03-05:
I am responding to the SVM-related questions in #63.
Closing this issue because original author did not reply any further.

closed 2019-03-05 by tlind