spencer-project/spencer_people_tracking #62: use of  'spencer_tracking_metrics'

CodeToPoem, 2018-12-13:
Hello,

I have a problem about how to use 'spencer_tracking_metrics' with data set. Can you provide a example for me？

Thanks for your help

Nat-zewge, 2019-02-23:
@CodeToPoem did you figure out how to use the metrics on a dataset?

tlind, 2019-02-23:
Essentially, you need to publish time-synchronized tracks and groundtruth tracks on the two topics listed in the main script. After no more data is published for several seconds, metrics will be computed and printed to the terminal. 
If I recall correctly, you should not publish/play your bagfiles too fast, as otherwise the Python-based (Approximate)TimeSynchronizer will run into a full buffer leading to a sort of deadlock.
Which part exactly is unclear to you?

tlind, 2019-02-23:
And there is an example launch file here:
https://github.com/spencer-project/spencer_people_tracking/blob/master/tracking/people/spencer_tracking_metrics/launch/clearmot_laser_detector.launch

CodeToPoem, 2019-02-24:
I still wonder that whether this project provided any dataset that contains visual data and groudtruth or not. @tlind

tlind, 2019-02-24:
We should have the short "MoCap" RGBD/2d laser dataset from our ICRA16 paper along with 3D groundtruth... let me find out where I uploaded it.
I am unfortunately not allowed to share the other datasets for privacy reasons.

CodeToPoem, 2019-02-26:
Thanks a lot.  @tlind 
And I am reading the example launch files in the "ICRA16" branch. I will try to adjust and run the launch file if I get the "MoCap" dataset.

FooKii, 2020-04-16:
And thanks for the package and framework, as a rookie, i would really appreciate a quick tutorial on spencer_tracking_metrics. 
So far my understanding is for [clearmot_from_logfile.launch](https://github.com/spencer-project/spencer_people_tracking/blob/master/tracking/people/spencer_tracking_metrics/launch/clearmot_from_logfile.launch) is:
* /ground truth from [srl_tracking_logfile_import](https://github.com/spencer-project/spencer_people_tracking/tree/master/tracking/people/srl_tracking_logfile_import), but the link for downloading CARMEN logfile is broken.
* detector running online, and result will be compared (evaluated) with the ground truth. 

As for another [clearmot_laser_detector.launch](https://github.com/spencer-project/spencer_people_tracking/blob/bcde8b78ea07f65920966a4d659ba8ef61aed9bc/tracking/people/spencer_tracking_metrics/launch/clearmot_laser_detector.launch):
* you can record your detection result in a bag file
* and your bag file will be evaluated with the groundtruth directly from pedsim.
* however, in code ```<remap from="/groundtruth" to="/tracks_with_simulated_occlusions"/>```, what does simulated occlusion mean? Does it mean that occlusion from pedsim will be removed as it's then impossible for detection from sensors, hence not count as false-positive if detector get it wrong.

So the requested Mocap data should be for the clearmot_from_logfile data, am i right?
if no Mocap, i should then use the second launch file.

Many Thanks