spencer-project/spencer_people_tracking #68: How can I use this packages with Hokuyo 2D laser sensor?

AssassinCrow, 2019-09-20:
I'm using the spencer_people_tracking packages with Asus Xtion pro for RGB-D camera (front camera only) and Hokuyo 2D lidar sensor for laser sensor(front laser only). To do this, I just modified name of parameter whose name is
</spencer/sensors/laser_front/echo0 >    to    < /scan > which is published topic's name from Hokuyo sensor's driver nodes for laser-sensored topic.
It works when I launch [laser_detectors.launch] only, but for [tracking_on_robot.launch], it just sends the same outputs when I launched with only one RGB-D camera.

this is the error i got
'''
[spencer/perception_internal/people_detection/rgbd_front_top/upper_body_detector-19] process has died [pid 17712, exit code -6,cmd /home/kane/spencer-people-tracking-ws/devel/lib/rwth_upper_body_detector/upper_body_detector __name:=upper_body_detector __log:=/home/kane/.ros/log/cb6a5b94-dae1-11e9-b8aa-004e019e642f/spencer-perception_internal-people_detection-rgbd_front_top-upper_body_detector-19.log].log file: /home/kane/.ros/log/cb6a5b94-dae1-11e9-b8aa-004e019e642f/spencer-perception_internal-people_detection-rgbd_front_top-upper_body_detector-19*.log
'''
and below is warning,
'''
Failed to lookup transform from frame laser into target frame base_footprint! Dropping input DetectedPersons message on topic /spencer/perception_internal/detected_persons/laser_front. Reason: "laser" passed to lookup Transform argument source_frame does not exist.
'''

Infact I'm using CUDA 9.0 version, but your team recommended CUDA 6.5, so could it be the reason why I can't get what I want??
If you can give me help, would you happened to explain how to solve this kinds of problem?  
Thanks in advance :)

tlind, 2019-09-26:
The upper-body detector is crashing, this could be for a variety of reasons. Try to enable logging to screen, or get a stack trace to see what is going on. I don't remember this component using CUDA at all, so it should not be a problem.

You also seem to be having some TF issue. Try to use tf_monitor, rqt etc. to debug connectivity of your TF graph. Maybe you need to add some static transform.

AssassinCrow, 2019-09-27:
Thanks for your reply. 
To be honest, in spencer page (https://github.com/spencer-project/spencer_people_tracking), I'm worried about this sentence 
' If no CUDA SDK is installed, the ROS package will still compile, but it will not provide any functionality.' at [Special note on CUDA SDK for the groundHOG detector].
I tried to compile cudaHOG library like the following manual page (https://github.com/spencer-project/spencer_people_tracking/tree/master/detection/monocular_detectors/3rd_party) but I failed in compiling.

( below is the error message I got)
'''
( All of these were done in the same directory.)
(make 'build' directory and I've done [$ sudo cmake]. )
$ cd /spencer-people-tracking-ws/src/spencer_people_tracking/detection/monocular_detectors/3rd_party
$ mkdir build && cd build
$ sudo cmake ..
$ export [my custom path of pkgconfig]
$ sudo make
.
.
.
Scanning dependencies of target libcudaHOG
[ 12%] Creating directories for 'libcudaHOG'
[ 25%] Performing download step (download, verify and extract) for 'libcudaHOG'
-- downloading...
     src='http://www.vision.rwth-aachen.de/media/resource_files/groundHOG_v1.1_b.tar.gz'
     dst='/home/robotics/spencer-people-tracking-ws/src/spencer_people_tracking/detection/monocular_detectors/3rd_party/build/libcudaHOG/src/groundHOG_v1.1_b.tar.gz'
     timeout='none'
-- [download 4% complete]
...
-- [download 96% complete]
-- [download 100% complete]
-- downloading... done
-- verifying file...
     file='/home/robotics/spencer-people-tracking-ws/src/spencer_people_tracking/detection/monocular_detectors/3rd_party/build/libcudaHOG/src/groundHOG_v1.1_b.tar.gz'
-- verifying file... warning: did not verify file - no URL_HASH specified?
-- extracting...
     src='/home/robotics/spencer-people-tracking-ws/src/spencer_people_tracking/detection/monocular_detectors/3rd_party/build/libcudaHOG/src/groundHOG_v1.1_b.tar.gz'
     dst='/home/robotics/spencer-people-tracking-ws/src/spencer_people_tracking/detection/monocular_detectors/3rd_party/build/libcudaHOG/src/libcudaHOG'
-- extracting... [tar xfz]
-- extracting... [analysis]
-- extracting... [rename]
-- extracting... [clean up]
-- extracting... done
[ 37%] Performing patch step for 'libcudaHOG'
patching file /home/robotics/spencer-people-tracking-ws/src/spencer_people_tracking/detection/monocular_detectors/3rd_party/build/libcudaHOG/src/libcudaHOG/cudaHOG/cudaHOG.cpp
[ 50%] No update step for 'libcudaHOG'
[ 62%] Performing configure step for 'libcudaHOG'
[ 75%] Performing build step for 'libcudaHOG'
WARNING: /home/robotics/spencer-people-tracking-ws/src/spencer_people_tracking/detection/monocular_detectors/3rd_party/build/libcudaHOG/src/libcudaHOG/cudaHOG/cudaHOG.pro:27: Unescaped backslashes are deprecated.
make[4]: nvcc: Command not found
Makefile:223: recipe for target 'gradients.o' failed
make[4]: *** [gradients.o] Error 127
Makefile:41: recipe for target 'sub-cudaHOG-make_default' failed
make[3]: *** [sub-cudaHOG-make_default] Error 2
CMakeFiles/libcudaHOG.dir/build.make:113: recipe for target 'libcudaHOG/src/libcudaHOG-stamp/libcudaHOG-build' failed
make[2]: *** [libcudaHOG/src/libcudaHOG-stamp/libcudaHOG-build] Error 2
CMakeFiles/Makefile2:67: recipe for target 'CMakeFiles/libcudaHOG.dir/all' failed
make[1]: *** [CMakeFiles/libcudaHOG.dir/all] Error 2
Makefile:127: recipe for target 'all' failed
make: *** [all] Error 2
'''

As you see, since I failed to compile cudaHOG library I couldn't but to skip this step.
So I think since I've never compiled cudaHOG library, this spencer packages do not support any functionality as written in [https://github.com/spencer-project/spencer_people_tracking](url) (at [Special note on CUDA SDK for the groundHOG detector]).

I still have CUDA 9.0 (also have nvidia driver)and have never compiled cudaHOG library for ground HOG detection. Could it be the reason why I got failed?

Thanks in advance :)

tlind, 2019-09-29:
The upper-body detector (which crashed in your initial error report) and groundHOG/libcudaHOG are two entirely different detectors. groundHOG/libcudaHOG/CUDA are _not_ required to run the upper-body detector, or a 2D laser detector.

(I believe you fail to compile libcudaHOG because you do not have nvcc on your $PATH. Check the README of the CUDA SDK on how to properly configure it after installation.)

AssassinCrow, 2019-10-01:
Thank you, I'm really relieved to hear that from the developer and sorry to bother you with the little things so many times.

After submitting the above question, I got progressed and eventually succeed in getting the upper body and laser scanned topics with spencer packages. 
Now I have two detected persons topics of the upper body and laser-scanned. 
As you know, the next step is to fuse these two topics to one robust data and I got the following error again.

`[ INFO] [1569930504.755381756]: New active topic registered: /spencer/perception_internal/detected_person_association/composite/lasers_upper_body_fused
[ INFO] [1569930504.755417309]: Number of active input topics has changed from 0 to 1. Re-configuring message filters. Active inputs for output topic '/spencer/perception_internal/detected_person_association/composite/lasers_upper_body_ground_hog_fused' are now: 
- /spencer/perception_internal/detected_person_association/composite/lasers_upper_body_fused

**[ WARN] [1569930506.481079391]: Occupancy grid map on topic /map timed out (or never received) by filter_detections_by_static_map! Will pass through all incoming detections until a map is received! Time offset was 1.56993e+09, but timeout is 1e+12 sec!**

[ INFO] [1569930507.519824035]: New active topic registered: /spencer/perception_internal/detected_person_association/composite/laser_rear
[ INFO] [1569930507.519871903]: Number of active input topics has changed from 1 to 2. Re-configuring message filters. Active inputs for output topic '/spencer/perception_internal/detected_person_association/composite/lasers_aggregated' are now: 
- /spencer/perception_internal/detected_person_association/composite/laser_front
- /spencer/perception_internal/detected_person_association/composite/laser_rear
`

In fact, I just simply added 
`<include file="$(find spencer_people_tracking_launch)/launch/detectors/laser_detectors.launch"/> `
to [tracking_single_rgbd_sensor.launch] and launch it.
(of couse I've laready modified the name of topics properly.)

When I launched tracking_single_rgbd_sensor.launch for the first time, it published tracked persons topic. I  think the reason was following; [tracking_single_rgbd_sensor.launch] do NOT fuse those two topics so it doesn't need the topic kind of map as a default.

Is it necessary to get a map file( generated by the g-mapping package with 2D-LiDar scanner for example) in advance?

Thanks in advance :)

closed 2019-10-07 by AssassinCrow

Alex-Beh, 2020-06-16:
@AssassinCrow Hello, I face some issue in compiling the libcudaHOG library. May I know your cuda version, operating system and nvidia-driver version?

Thanks in advance.

AssassinCrow, 2020-06-16:
Actually I removed the ubuntu os, by my mistake... And it's been a year from I used this package lastly, so I'm not sure how it worked.
But I surely remember that I also faced some error wrt compile. To begin with the conclusion, you would not need to care about HOG error. All I did were just to modify only launch files (never cpp source codes) so that the corresponded topic can be processed.
Just you know, my condition is

OS: Ubuntu 16.04 
ROS: ROS kinetic
RGB-D camera: Asus Xtion camera 
LiDAR: Hokuyo 2D LiDAR
Nvidia Driver: 384   <--- I'm not sure
CUDA: CUDA 9.0
cuDNN: ... <--- also, not sure  :)

maybe, you don't need to use cuDNN.