spencer-project/spencer_people_tracking #69: pcl running error: Input data has been check, but no people is detected

Usthiel, 2019-09-25:
Hi,

I follow your tutorials to use Asus Mini single camera to test people detection. The upper-body detection runs good with run the tracking_single_rgbd_sensor.launch. But PCL people detection is not working.  I have been set 'use_upper_body_detector' to 'false', and 'use_pcl_detector' to true as the tutorial mentioned. 
But once I run this detector,  no person is detected. I have been check the input topic data (pointcloud and camera_info) is not empty, and topic (/spencer/perception/detected_persons) publish data even if the person array is empty. Sometimes the terminal print two lines of error like below:
[pcl::KdTreeFLANN::setInputCloud] Cannot create a KDTree with an empty input cloud!
[pcl::people::pcl::people::HeadBasedSubclustering::subcluster] Cluster indices have not been set! 
Have no idea why the fully-body detector never run. rviz shows the pointcloud and tf correctly but no detected person box exists. Does there are some params should be changed? By the way, I put my camera 1.1 meters above the ground, not sure whether it is not suitable. If you have any ideas about this error, please give me some advices. 

Thank you!

tlind, 2019-09-26:
Hi,
in my experience, this error mostly occurs when something is wrong with your ground plane. Maybe your base_link is pointing into the wrong direction (behind the camera, instead of forward)?

1.1m itself should be fine. Try to play around a bit with the ground-plane coefficients (e.g. try negative last element), or add a static TF transform that rotates your base_link by 180 deg in yaw. There are also some debug visualizations, e.g. a g_markerPublisher which visualizes the ground plane, as well as a point cloud of all points above ground. For the former, the ground plane visualization should lie in front of your RGB-D sensor.

Usthiel, 2019-10-07:
Hi, Sorry for the late reply, thanks for your advice, it seems that there is no problem now.

closed 2019-10-07 by Usthiel

FooKii, 2020-04-15:
Hi, i got the same problem. One more thing i noticed:
* pcl_detector publishes 2 topics : /pcl_detector/ground_plane[point_cloud2] & /no_ground_plane [pointcloud2 type].
* when "Cannot create a KDTree with an empty input cloud!" occurs, /ground_plane publish no pointcloud data. 

After playing with base_link and ground_height setting(add yaw to base_link and height to 1 m), i have data from both topics and no more error.   
But still no detection result. I'm running in gazebo simulation, i'll try again when i work with the real robot, maybe i'll give some positive updates later.

Would also be great if you can give me some hints too,
Thanks

tlind, 2020-04-15:
This is easiest to debug if you switch on the visualization inside the detector, and then look at the published markers in Rviz. The ground plane marker should lie in front of the sensor (not behind it) and at the height and orientation at which you would expect your groundplane to be. Feel free to share a screenshot if you need further help.

FooKii, 2020-04-15:
Thanks for the reply, here is the screenshot, could it be the reason that only a small area of ground is detected?
Also i notice that i still get the "can't create KDtree" error at the beginning, then it stops after i add the human model in front the robot.

Thanks
[image]

tlind, 2020-04-16:
The visualization topic you are showing is for the upper-body detector. However, the problem you describe with the KDL tree relates to the PCL detector. These are two different detectors, which can be selected using the arguments described in the original question (first post).

Instead, for the PCL detector, try to visualize in Rviz the "misc_marker_array" topic (using a MarkerArray display), and the "ground_cloud" and "no_ground_cloud" topics (PointCloud2 displays). All three topics should be published by the PCL detector.

FooKii, 2020-04-16:
Hey, 
seems i find a problem, there's nothing in rviz for  "ground_cloud" and "no_ground_cloud" topics, but i can see data published by them using rostopic echo. Also misc_marker_array will only be shown as a darker area on the ground in front of the robot. 
Maybe you have an idea on this problem?
thanks
[image]
[image]