spencer-project/spencer_people_tracking #74: ROS Melodic support

NunoGuedesFollow, 2020-03-26:
Is there going to be a ROS Melodic support?

AssassinCrow, 2020-04-05:
Hello, I'm not the developer but I'm also trying to execute the `spencer` package in ROS melodic. 
Maybe you can find your answer in [here](https://github.com/LCAS/rosdistro/wiki#using-the-l-cas-repository-if-you-just-want-to-use-our-software)(it says `spencer` package also supports ROS melodic.)

NunoGuedesFollow, 2020-04-06:
Thanks, I will try it!

Best Regards!

tlind, 2020-04-10:
Hi, adding ROS Melodic support is on our roadmap. I know there are some issues related to C++11 (namespacing issues with boost::shared_ptr and std::shared_ptr, e.g. in the rviz plugins) which currently prevent compilation also on the LCAS build farm for ROS Melodic.

However, I cannot give an ETA right now since I am mostly still using ROS Kinetic and first have to set up a virtual machine for testing. Pull requests are also welcome!

NunoGuedesFollow, 2020-05-27:
Hi,

Have you had any success by now in this matter?

Best Regards,
Nuno Guedes

tlind, 2020-05-30:
No progress on this matter so far from my side due to other duties. It seems we will skip Melodic and directly switch to Noetic / Ubuntu 20.04 which will be the final ROS 1 release.

Alex-Beh, 2020-06-12:
@NunoGuedesFollow May I know have you compile the code successfully in Melodic? I am trying to do the migration, currently facing some issue on the opencv3

tlind, 2020-06-12:
I haven't tested it myself yet, but using the method described in the link, it should be possible (assuming OpenCV 2 is still installable under your OS version).

Alex-Beh, 2020-06-13:
@tlind Hello, may I know which version you using for the opencv? 2.4.9?

Alex-Beh, 2020-06-23:
@tlind Sorry for interupt you again. I am still facing some issues.

There is no publisher to topic `rgb/image_rect_color` , may I know is this normal?
[image]

I get the detection result, but there is no tracking functionality. Is this because I am too near to the camera or something is wrong?

tlind, 2020-06-23:
The names of the topics can vary a bit depending on which sensor and OpenNi version you use. Maybe there is "rgb/image_color_rect" or "rgb/image_color/raw" which you can use instead?

If you get detection results, but no tracking results, it is most likely due to an issue with the TF "odom" frame, which is required for tracking. Run "rosrun tf view_frames" and check that your sensor frame has a connection to odom frame.

Alex-Beh, 2020-06-23:
@tlind I am using the realsense D435 with the realsense-ros driver. I think the detector do not subscribe to the correct image topic. I have both the following topics:
```
/spencer/sensors/rgbd_front_top/color/image_raw
/spencer/sensors/rgbd_front_top/rgb/image_rect_color
```
I tried `rostopic info /spencer/sensors/rgbd_front_top/rgb/image_rect_color`
```
Type: sensor_msgs/Image

Publishers: 
 * /spencer/sensors/rgbd_front_top/realsense2_camera_manager (http://localhost:41119/)

Subscribers: 
 * /spencer/sensors/rgbd_front_top/realsense2_camera_manager (http://localhost:41119/)
 * /spencer/perception_internal/people_detection/rgbd_front_top/upper_body_detector (http://localhost:40693/)
```
However, when I `rostopic echo /spencer/sensors/rgbd_front_top/rgb/image_rect_color`, there is no message from the topic. May I know which topic should be subscriber by the detector?

From the tf frame, the `rgbd_front_top_link` and `odom` are connected although the depth frame, infra1 & infra2 do not update periodically. (The buffer length is always 0sec and I think that it is normal because they are fixed link relationships)
[frames.pdf](https://github.com/spencer-project/spencer_people_tracking/files/4819374/frames.pdf)

Alex-Beh, 2020-06-26:
@tlind Sorry for keep asking question. Have been struggled for few weeks.

Do the rosbag file come with tracking functionality? I try the following command:
`roslaunch spencer_people_tracking_launch tracking_on_bagfile.launch
`

and below is the screenshot from rviz. It seems there is no tracking function.
[image]

I try ping the `/spencer/perception/detected_persons` and there is no data come out. 
```
alex-beh@alexbeh-ThinkPad-T490:~$ rostopic hz /spencer/perception/detected_persons
subscribed to [/spencer/perception/detected_persons]
WARNING: may be using simulated time
no new messages
no new messages
no new messages
no new messages
no new messages
```

Looking forwards to your help. Thanks

Alex-Beh, 2020-08-28:
The above error is about aggregate error and I think you have solved the issue. I try the melodic brnach on my laptop with the tracking_on_bagfile.launch. Although there are tracking persons shown in the rviz, but seem the groundHOG detector is not running. 

I try `rosnode list |grep groundHog` and `rostopic list|grep groundHOG` but there is no results. May I know the groundHOG detector is running on your system?

tlind, 2020-08-28:
Remember you first need to install the CUDA Toolkit (e.g. 10.1) and then manually make and install libcudaHOG (as described in `detection/monocular_detectors/3rd_party/README.md`), before re-building the `rwth_ground_hog` catkin package. Without the libcudaHOG compile-time dependency, the detector will be non-functional and just print a red error message to the console upon startup.

However, indeed some more patching was required to make libcudaHOG compile under Ubuntu 18.04. The commit which I just pushed to the melodic branch (1970b37f7) provides the necessary fixes. I just tested it in a Docker container with `tracking_on_bagfile.launch use_hog_detector:=true`. If you switch on the DetectedPersons visualization for HOG, you should see yellow detection boxes. I also verified that the PCL detector (`use_pcl_detector:=true`) still works.

tlind, 2020-08-28:
Here's how it should look like then (note that in this rather confined space, the groundHOG does not output many detections as it is intended for far-range):

[image]

Alex-Beh, 2020-08-29:
I installed Cuda 10.1 and check the installation with nvcc --version and using gcc-6 & g++-6 to compile the library.
```
nvcc: NVIDIA (R) Cuda compiler driver
Copyright (c) 2005-2019 NVIDIA Corporation
Built on Sun_Jul_28_19:07:16_PDT_2019
Cuda compilation tools, release 10.1, V10.1.243
```

However, when I try to install the libcudaHOG library and I encounter the following issue when I make:
> undefined reference to `boost::program_options::options_description::options_description

I think the boost library didn't link to the makefile correctly. I saw you have uploaded a bash file `fix_boost_dependency.sh` and I tried run the file and

> Patching boost program options dependency recursively in /home/alex-beh
> grep: src/: No such file or directory
> sed: no input files

May I know do you have any idea on this?

Alex-Beh, 2020-08-30:
Yes, the issue has been solved by reinstall the groundHOG library. However, the accuracy of the model decreased after the changes.

I saw you have retrained the model for leg detector with laser. May I know how you retrain the config file for the upper_body_detector or only reformat the file?

You can see the results of the previous version in the following link:
https://drive.google.com/file/d/1WpgNIhiJXu_20S_vKd8nJW4A1rYd3cYP/view?usp=sharing

I will do the comparison with the current branch and upload the result tomorrow.

tlind, 2020-08-30:
It would be helpful to know if you think it to be related to 2D laser, or RGB-D.

For 2D laser, you can try to switch the model (specified via [model_prefix](https://github.com/spencer-project/spencer_people_tracking/blob/c63cd4622528cb8e6308fc08994b7578abcfe51c/detection/laser_detectors/srl_laser_detectors/launch/learned_detector_with_segmentation.launch#L12)) from the lms500 to lms200 model, which used to be the default before. Unfortunately I do not recall on which dataset the original lms200 detector been trained (Freiburg main station, aspekt, or a combination thereof?), thus some difference in performance could be expected.

For the upper-body detector, I believe PR #59 from LCAS/RWTH introduced mostly a change in file format to allow easier use of the ROS parameter server, but I'm not 100% sure. I'll try to convert the template back into the old format that you can visualize nicely e.g. using VSCode (you will recognize the silhouette as "ASCII art").

germal, 2020-11-27:
Hello @tlind , I'm trying to build Spencer on ROS melodic , but catkin throws errors related to Opencv3 compatibility in person_classifier.hpp 
also if I changed the namespaces gpu->cuda and HOGdescriptor->HOG , for example 

error: ‘getPeopleDetector64x128’ is not a member of ‘cv::cuda::HOG’
     setSVM(128, 64, cv::cuda::HOG::getPeopleDetector64x128(), 0.0);

could you please advice how to resolve ?
Thanks a lot,
germal

germal, 2020-11-27:
Hello @Alex-Beh !
I have built Spencer on melodic and I'd like to adapt to D435 .Did you managed to make it work with Spencer ? could you advice what to change ?
Thanks a lot !
Regards
germal