spencer-project/spencer_people_tracking #78: Nothing shown when I try tutorial 1 with rosbag file

Alex-Beh, 2020-06-17:
Hello, I would like to the tutorial 1 with the rosbag file.

The following two commands have been executed:
`roslaunch spencer_people_tracking_launch tracking_single_rgbd_sensor.launch height_above_ground:=1.6`
`rosbag play front_rgbd.bag tf_odom.bag 
`

Is this the correct bag file? In troubleshooting([url](https://github.com/spencer-project/spencer_people_tracking#troubleshooting)), it mentioned to check whether the pointcloud is display properly. But there is nothing shown in RVIZ. In rviz, the RGB-D cloud subscribed to `/spencer/sensors/rgbd_front_top/depth_registered/points`.

Besides, what is the fixed frame should I choose? There is `Warning: TF_OLD_DATA ` from the terminal where I roslaunch the file.

Thanks in advance. Cannot wait to see the amazing result.

Alex-Beh, 2020-07-15:
Fixed by changing `subscriber.getSubscriber().getNumPublishers()` to `subscriber.getSubscriber()`.

And no empty messages in input_topics that been subscribe by the aggregate nodelet.

For example, I am not using any rear sensor. So I need to comment out the `composite/rgbd_ground_hog_rear`  in the input_topics.

```
<!-- Aggregate groundHOG in RGB-D -->
        <node name="aggregate_rgbd_ground_hog" pkg="nodelet" type="nodelet" args="load spencer_detected_person_association/AggregateDetectionsNodelet detected_person_association_nodelet_manager" output="screen">
            <rosparam param="input_topics">
                - composite/rgbd_ground_hog_front
                <!-- - composite/rgbd_ground_hog_rear -->
            </rosparam>
            <remap from="output" to="composite/rgbd_ground_hog_aggregated"/>
        </node>
```

tlind, 2020-08-05:
What you describe in your first comment is by design (if there is no detector publishing any messages, the aggregator nodelets should also not publish anything). However, you have to make sure that you are not running any detector (advertising such detection topics) for sensors which don't exist. Therefore, the `tracking_on_bagfile.launch` exposes all these arguments such as `rear_laser` etc. which you need to set to `false`.

I don't fully understand what this change implies

> Fixed by changing subscriber.getSubscriber().getNumPublishers() to subscriber.getSubscriber().

and what  you mean by

> And no empty messages in input_topics that been subscribe by the aggregate nodelet.

I agree though that the current detection-fusion pipeline with all these converter nodelets is a bit intransparent and difficult to debug. Theoretically, it might be easier to fuse all detections from different sensors right inside the tracker without any nodelets in between.

tlind, 2020-08-12:
I investigated the issue a bit further and it seems that the laser detector crashes under certain ROS / Ubuntu versions upon receipt of the first laser scan due to changes in Eigen and OpenCV. Normally, if the laser detector is not running, the convert_to_composite nodelet would not advertise any output topic (and thus not block the aggregator nodelet). However, in this case, the crash leads to the ROS master being unaware of the detector node being shutdown. I'm working on a fix.

Alex-Beh, 2020-08-12:
Thanks for that. I didn't notice the clash issue, so I bypass the checking statement which is not a proper solution. 

> **Fixed** by changing `subscriber.getSubscriber().getNumPublishers()` to `subscriber.getSubscriber()`.

I am currently trying this package successfully on Melodic rosdistro and going to migrate to Noetic. May I know are you going to release the Noetic version? Do you have any estimated timeline in mind?

tlind, 2020-08-12:
I've got a working version for Melodic ready and it seems there are not so many changes required for Noetic, so I'd say rather sooner than later. Couple of days maybe.

Alex-Beh, 2020-12-08:
@tlind Thanks. The latest branch solved the previous issue. However, I realized the track id with missing detection at certain frame will converge towards its nearest neighbour (which is a laser data from static obstacle and it caused high number of ID switches.) I try tune the `max_gating_distance` in nnt.launch , it do help but the issue still remains. Which parameters should I do the tuning?

tlind, 2020-12-09:
There are several parameters (including the motion model, process noise etc.) which have an impact on data association. The most important one is the measurement noise, which in the standard configuration is read from the covariance matrix in the DetectedPersons message. Only the first 2 rows and columns (x and y position over the ground plane) are relevant for tracking. You could e.g. set the first two diagonal elements to a lower value, in order to express a higher certainty and thus be more strict in data association.

An alternative, simpler way is to completely overwrite the measurement noise after measurements have arrived inside the tracker. See the following two parameters:
https://github.com/spencer-project/spencer_people_tracking/blob/09b256ba4bc22c5cae8a5ae88960de1a387cfd7f/tracking/people/srl_nearest_neighbor_tracker/launch/nnt.launch#L68-L69

However, if you set the noise level too low, data association might always fail, resulting in a very high number of ID switches. The trade-off can be difficult to get right for all kinds of scenarios.

Alex-Beh, 2020-12-09:
@tlind Then what is the trade-off if set the measurement noise level too high?  Will include more uncertainty into the tracking process?

The motion model and process noise that you mentioned are those in the imm parameter files? I saw there are few imm parameter file with different combinations of motion models inside the tracking folder. May I know how to choose the motion models, markov transition probabilites and the cosvxx/cosvyy for each models?