spencer-project/spencer_people_tracking #85: Intergrate other detection algorithm.

lff12940, 2020-12-15:
I integrated other detection algorithms and used the provided scripts for message conversion, but I encountered the following problems:

[ WARN] [1608023759.960176417, 1439557604.073015311]: Jump back in time detected. Deleting all existing tracks. Old time was 1439557603.966, new time is 1439557603.945 (dt=-0.021 sec)!
[ WARN] [1608023760.593699488, 1439557604.198680538]: Jump back in time detected. Deleting all existing tracks. Old time was 1439557604.099, new time is 1439557604.057 (dt=-0.042 sec)!
[ WARN] [1608023761.217882125, 1439557604.323381596]: Jump back in time detected. Deleting all existing tracks. Old time was 1439557604.233, new time is 1439557604.138 (dt=-0.095 sec)!

The detection algorithm is: DROW and DR-SPAAM person detectors[https://github.com/VisualComputingInstitute/DR-SPAAM-Detector](url)

Can anyone help me? How to solve this problem?

THanks.

tlind, 2020-12-15:
Hi, it seems that your messages arrive slightly out-of-order. An easy workaround would be to add a small tolerance in delta time in the condition of the if statement:
https://github.com/spencer-project/spencer_people_tracking/blob/09b256ba4bc22c5cae8a5ae88960de1a387cfd7f/tracking/people/srl_nearest_neighbor_tracker/src/srl_nearest_neighbor_tracker/nearest_neighbor_tracker.cpp#L227-L230

Are you running multiple detectors in parallel (all publishing to the detected persons topic that is input to the tracker), and they are not temporally synchronized?

lff12940, 2020-12-16:
Thanks. It worked when I add a tolerance in delta time.

I replaced the laser detectors with DR-SPAAM-Detector in the launch file tracking_on_bagfile.launch.
At present, I have integrated DR-SPAAM detector and upper-body detector using front_laser and front_rgbd, but the fused detections are not good. There are many false detections in the test results. How should I fine tune for better result?

Thanks.

tlind, 2020-12-21:
Which detector is mostly causing these false alarms? Or are the detections from the individual detectors okay, and the problem is at the fusion stage? Maybe you could attach a picture or short video.