spencer-project/spencer_people_tracking #89: python listener

jonathanleinola, 2021-03-05:
Hey,
could you point me out how to do a python listener for the matched object? I could not find for how to listen for eg. this topic: /spencer/perception/tracked_persons/tracks. Tried with: rospy.Subscriber('/spencer/perption/tracked_persons/tracks', Pose, callback), but i think it's not right.

Also, can i somehow extract the velocity parameters too?

Thank you

tlind, 2021-03-06:
You need to use `TrackedPersons' as the second argument. And import it:

    from spencer_tracking_msgs.msg import TrackedPersons, TrackedPerson

You can get the estimated linear velocity using trackedPerson.twist.twist.linear.x and .y

jonathanleinola, 2021-04-16:
Hi, is it possible to get the trackedperson in map frame too somehow? if I have a camera mounted on top of a robot?

closed 2021-05-09 by jonathanleinola