spencer-project/spencer_people_tracking #90: Get detected person coordinates in a map coprdinate system

jonathanleinola, 2021-03-07:
Hi,

Is it somehow possible to get the detected person coordinates in map coordinates? if i switch the dummy transformation to in relate to base_link, the coordinates of the detected person is still at in relation to the camera coordinates. 

Greetings,
Jonathan

jonathanleinola, 2021-03-30:
I am able to link it with the robot's baselink. the positions of tracked person seems right, but the tracking get affected by the frequency of the published baselink topic. any help?

tlind, 2021-03-30:
The tracker always has to wait for the transform from odom to base_link to arrive, before it can transform the detections into the fixed world frame. You could increase the publish frequency by interpolating the transforms, or dating the timestamp of the published transform a bit into the future so the tracker does not have to wait for it to arrive.

jonathanleinola, 2021-03-30:
Oh, interesting. i will take a look on that. What if i connect them both to robot's base_link? the base_footprint_id and world_frame_id. i seem not understand the difference of these frames

closed 2021-05-09 by jonathanleinola