spencer-project/spencer_people_tracking #96: People tracking process dies

jonathanleinola, 2021-04-11:
Hi, 

Do you have an idea why the people tracking module crashes when the detection from the upper body is lost? And what could i do with it. Can i somehow reset the tracking module?
Error:
[spencer/perception_internal/people_tracking/post_processing/filter_tracks_confirmed_by_upper_body-36] process has died [pid 4515, exit code -11]

Alex-Beh, 2021-04-14:
The tracking process didn't die. Only the filter module die because of some bug inside the code. You can refer to [this commit](https://github.com/PhiAbs/spencer_people_tracking_yolo/commit/fa543f0649e5388884cc95e2dafa92d0dd6f073a).

jonathanleinola, 2021-04-16:
Awesome, thanks for pointing me in the right direction. I'll go through it and see if I can find it

kaushikbalasundar, 2021-05-04:
@jonathanleinola were you able to solve this bug?

closed 2021-05-09 by jonathanleinola