spencer-project/spencer_people_tracking #98: Tracker does not work with static people

bach05, 2021-06-03:
Hi, 
I'm using the framework in ROS Melodic with the PCL-based detector. I would like to know why the tracker is not working with static people. In the detection topic I have both static and moving people: 

```
header: 
  seq: 424
  stamp: 
    secs: 126
    nsecs: 563000000
  frame_id: "base_footprint"
detections: 
  - 
    detection_id: 804
    confidence: 0.789739012718
    pose: 
      pose: 
        position: 
          x: 2.51309752464
          y: 0.268414497375
          z: 0.833835959435
        orientation: 
          x: 1.0
          y: 0.0
          z: 0.0
          w: 0.0
      covariance: [0.028900000000000006, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.028900000000000006, 1.4248800100554528e-33, 0.0, 0.0, 0.0, 0.0, 1.4248800100554528e-33, 0.028900000000000006, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 999999.0, 999999.0, 999999.0, 0.0, 0.0, 0.0, 999999.0, 999999.0, 999999.0, 0.0, 0.0, 0.0, 999999.0, 999999.0, 999999.0]
    modality: "rgbd"
  - 
    detection_id: 805
    confidence: 0.64065861702
    pose: 
      pose: 
        position: 
          x: 2.38973236084
          y: 1.2305482626
          z: 0.830446779728
        orientation: 
          x: 1.0
          y: 0.0
          z: 0.0
          w: 0.0
      covariance: [0.028900000000000006, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.028900000000000006, 1.4248800100554528e-33, 0.0, 0.0, 0.0, 0.0, 1.4248800100554528e-33, 0.028900000000000006, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 999999.0, 999999.0, 999999.0, 0.0, 0.0, 0.0, 999999.0, 999999.0, 999999.0, 0.0, 0.0, 0.0, 999999.0, 999999.0, 999999.0]
    modality: "rgbd"
---
header: 
  seq: 425
  stamp: 
    secs: 126
    nsecs: 633000000
  frame_id: "base_footprint"
detections: 
  - 
    detection_id: 806
    confidence: 0.779071986675
    pose: 
      pose: 
        position: 
          x: 2.51674199104
          y: 0.270926356316
          z: 0.833391308784
        orientation: 
          x: 1.0
          y: 0.0
          z: 0.0
          w: 0.0
      covariance: [0.028900000000000006, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.028900000000000006, 1.4248800100554528e-33, 0.0, 0.0, 0.0, 0.0, 1.4248800100554528e-33, 0.028900000000000006, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 999999.0, 999999.0, 999999.0, 0.0, 0.0, 0.0, 999999.0, 999999.0, 999999.0, 0.0, 0.0, 0.0, 999999.0, 999999.0, 999999.0]
    modality: "rgbd"
  - 
    detection_id: 807
    confidence: 0.662306666374
    pose: 
      pose: 
        position: 
          x: 2.38957381248
          y: 1.22992503643
          z: 0.831639051437
        orientation: 
          x: 1.0
          y: 0.0
          z: 0.0
          w: 0.0
      covariance: [0.028900000000000006, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.028900000000000006, 1.4248800100554528e-33, 0.0, 0.0, 0.0, 0.0, 1.4248800100554528e-33, 0.028900000000000006, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 999999.0, 999999.0, 999999.0, 0.0, 0.0, 0.0, 999999.0, 999999.0, 999999.0, 0.0, 0.0, 0.0, 999999.0, 999999.0, 999999.0]
    modality: "rgbd"
---
```

But in the tracked topic I have only moving people: 

```
header: 
  seq: 276
  stamp: 
    secs: 126
    nsecs: 633000000
  frame_id: "odom"
tracks: 
  - 
    track_id: 0
    is_occluded: False
    is_matched: False
    detection_id: 0
    age: 
      secs: 4
      nsecs: 650000000
    pose: 
      pose: 
        position: 
          x: -5.56138168039
          y: -5.39720909843
          z: 0.0
        orientation: 
          x: 0.0
          y: 0.0
          z: -0.982902285511
          w: 0.184127936873
      covariance: [0.06664105965826517, -0.018896863912598684, 0.0, 0.0, 0.0, 0.0, -0.01889686391259868, 0.09972256787969144, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 9999999.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 9999999.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 9999999.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.030461741978670857]
    twist: 
      twist: 
        linear: 
          x: -0.64919326339
          y: -0.25207386438
          z: 0.0
        angular: 
          x: 0.0
          y: 0.0
          z: 0.0
      covariance: [0.16929918165952346, -0.06736934552562276, 0.0, 0.0, 0.0, 0.0, -0.06736934552562276, 0.3075734284549735, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 9999999.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 9999999.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 9999999.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 9999999.0]
---
```

 I attach also my launch file. Am I doing something wrong or what? 

[tracking_single_rgbd_sensor_personalized.txt](https://github.com/spencer-project/spencer_people_tracking/files/6590944/tracking_single_rgbd_sensor_personalized.txt)
[front_rgbd_detectors_personalized.txt](https://github.com/spencer-project/spencer_people_tracking/files/6590948/front_rgbd_detectors_personalized.txt)
[freiburg_people_tracking_personalized.txt](https://github.com/spencer-project/spencer_people_tracking/files/6590952/freiburg_people_tracking_personalized.txt)
[start_personalized.txt](https://github.com/spencer-project/spencer_people_tracking/files/6590954/start_personalized.txt)

Thank you in advance!

tlind, 2021-06-07:
In freiburg_people_tracking_personalized.txt, when you include `nnt.launch`, try to set its argument `use_initiation_logic` to `false`. The initiation logic, with the default set of parameters, requires some motion for tracks to be initiated.

bach05, 2021-06-08:
Ok, thank you! It worked. 

By the way, I may have some other questions in other aspects of the framework. Could I open new issues?

closed 2021-09-08 by bach05